robotology/mesh-support #467: The robot remains position controlled even after pressing fault button

gabrielenava, 2017-05-22:
### Description of the failure

I pressed the fault button, but the robot remains position controlled. Even after removing the cable, the robot was still controlled. As you can see in the photo below, the fault is now useless:

[image]

S-Dafarra, 2017-05-22:
[dumpLogger.txt](https://github.com/robotology/icub-support/files/1018422/dumpLogger.txt)

julijenv, 2017-05-22:
Hi @gabrielenava  and @S-Dafarra ,
with the help of @MrAndrea, we debug the issue - the fault was touching a piece of the main robot structure.
everything should be working fine now.

closed 2017-05-22 by julijenv