robotology/mesh-support #468: Weird behaviour of right wrist prosup

gabrielenava, 2017-05-26:
### Description of the failure
Sometimes, it happens that slightly moving the right wrist prosup causes that joint to go in "not configured" status. This also happened while the robot was torque controlled: as a result, the leg joints went in hardware fault. I will update the issue with the logger output as soon as it will happen again.

gabrielenava, 2017-05-29:
UPDATE: it happened again: this time the robot was position controlled. After touching the right arm, most of the joints went in hardware fault status, and the arms in NOT CONFIGURED status. This is the error in the logger:

```
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j12_15 IP 10.0.1.29 for nv ID32 = 0x0100010e -> IND = 1, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j=1) for BOARD rightArm-j12_15 IP 10.0.1.29
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j12_15 IP 10.0.1.29 for nv ID32 = 0x0100010b -> IND = 1, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD rightArm-j12_15 IP 10.0.1.29 joint  1 
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j12_15 IP 10.0.1.29 for nv ID32 = 0x0100010b -> IND = 1, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getTargetPositionRaw() could not read reference pos for  BOARD rightArm-j12_15 IP 10.0.1.29 joint  1 
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j12_15 IP 10.0.1.29 for nv ID32 = 0x0100020e -> IND = 2, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j=2) for BOARD rightArm-j12_15 IP 10.0.1.29
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j12_15 IP 10.0.1.29 for nv ID32 = 0x0100020b -> IND = 2, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD rightArm-j12_15 IP 10.0.1.29 joint  2 
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j12_15 IP 10.0.1.29 for nv ID32 = 0x0100020b -> IND = 2, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getTargetPositionRaw() could not read reference pos for  BOARD rightArm-j12_15 IP 10.0.1.29 joint  2 
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j12_15 IP 10.0.1.29 for nv ID32 = 0x0100030e -> IND = 3, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j=3) for BOARD rightArm-j12_15 IP 10.0.1.29
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j12_15 IP 10.0.1.29 for nv ID32 = 0x0100030b -> IND = 3, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD rightArm-j12_15 IP 10.0.1.29 joint  3 
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j12_15 IP 10.0.1.29 for nv ID32 = 0x0100030b -> IND = 3, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getTargetPositionRaw() could not read reference pos for  BOARD rightArm-j12_15 IP 10.0.1.29 joint  3 
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j0_3 IP 10.0.1.3 for nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j=0) for BOARD rightArm-j0_3 IP 10.0.1.3
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j0_3 IP 10.0.1.3 for nv ID32 = 0x0100000b -> IND = 0, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD rightArm-j0_3 IP 10.0.1.3 joint  0 
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j0_3 IP 10.0.1.3 for nv ID32 = 0x0100000b -> IND = 0, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getTargetPositionRaw() could not read reference pos for  BOARD rightArm-j0_3 IP 10.0.1.3 joint  0 
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j0_3 IP 10.0.1.3 for nv ID32 = 0x0100010e -> IND = 1, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j=1) for BOARD rightArm-j0_3 IP 10.0.1.3
[... cut ...]
[ERROR]embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD rightArm-j8_11 IP 10.0.1.28 joint  3 
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD rightArm-j8_11 IP 10.0.1.28 for nv ID32 = 0x0100030b -> IND = 3, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getTargetPositionRaw() could not read reference pos for  BOARD rightArm-j8_11 IP 10.0.1.28 joint  3 
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
```

gabrielenava, 2017-05-29:
cc @julijenv  @spada1978

marcoaccame, 2017-05-31:
What I see with this fragment of log file is that the boards on the right arm do not respond to requests over UDP. The boards are 10.0.1.29 (rightArm-j12_15), 10.0.1.3 (rightArm-j0_3), 10.0.1.27 (rightArm-j4_7), 10.0.1.28 (rightArm-j8_11).

I need the complete log to understand why that happens.

My guess is that one or more boards have reset and restarted due to a short circuit after the touch.

Please, install latest FW version of ETH boards (dated 29 may 17) and pull icub-main to get benefit from extra diagnostics.

Then, please call me ASAP after it happens.

S-Dafarra, 2017-06-06:
This issue seems related to https://github.com/robotology/icub-support/issues/439 . Indeed it seems that some boards in the right arm go down when touched or when performing some fast movements. Yesterday it happened again, and the motors led in the backpack was blinking. The log is attached. 
[log_icub-head_yarprobotinterface_6521_5june.txt](https://github.com/robotology/icub-support/files/1054056/log_icub-head_yarprobotinterface_6521_5june.txt)

marcoaccame, 2017-06-06:
The log shows that the board 10.0.1.3 loses connection with what is after it: 10.0.1.28 etc. (right arm, torso, legs) for 2 seconds at first, then indefinitely. 

```
from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 478s 590m 398u: 
(code 0x06000001, par16 0x0001 par64 0x0000000000000000) -> ETH monitor: link goes down ...
... errors
from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 480s 691m 397u: 
(code 0x06000000, par16 0x0001 par64 0x0000000000000000) -> ETH monitor: link goes up ...
... errors
from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 483s 890m 398u: 
(code 0x06000001, par16 0x0001 par64 0x0000000000000000) -> ETH monitor: link goes down
... errors
```

You have to look with FirmwareUpdater program the status of the boards 10.0.1.28 and beyond.

If they are in IDLE, then they restarted and it is a power supply problem. 
If they are still in RUNNING mode, then it is a loose ETH cable in output port P3 of board 10,0.1.3.

francesco-romano, 2017-06-26:
This issue is still present and quite problematic.
This is reproducible by a joint going in hardware fault. When it is put in idle, then the joint goes in not-configured and the `yarprobotinterface` should be restarted.
Given that the HW Fault can happen for a lot of reasons, e.g. current limits, restarting every time the robot is not feasible.

@marcoaccame @valegagge @randaz81

valegagge, 2017-06-26:
Hi @francesco-romano ,
this behaviour you descibed is not a "tipical" behaviour of firmware when an hardwrae fault happen: it is due to a particular problem on right arm of green robot. 

In order to understand what is going on, it is necessary to know if applications, running on right arm boards, are in IDLE or RUNNING , like @marcoaccame wrote in previw post.

If you don't know how to get this information, I can explain you.

valegagge, 2017-06-26:
@francesco-romano and I reproduced this issue moving prosupination joint by hand and we verified that after only the application of  board eb27 is in IDLE status.
We made the same procedure on left arm, but the behaviour is different: we need to apply more force before robot goes in protection state.

Therefore, we presume there is some electronic issue on that board/body part.

@julijenv : could you check if all cabling is ok on right arm? Maybe we can start to change mc4plus board....

marcoaccame, 2017-07-20:
Following the explanation of @valegagge and test done together with @francesco-romano, I assume that it is not a bug anymore, hence I remove the tag `bug`.

Moreover, if really the board `EB27` goes from `RUNNING` to `IDLE` status after moving by hand the prosupination joint it means that it has had a reset, hence there is some cabling or HW issue to be solved.