robotology/mesh-support #471: Three right hand fingers remain bended after startup

gabrielenava, 2017-05-26:
### Description of the failure
The first three fingers of the right hand remain bended after finishing the startup procedure. The problem seems to be related to this error:

```
<#STRING_START#>HostTransceiver::addSetMessage__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  15 0 656 
<#STRING_END#>
20220
48,881958
4
<#STRING_START#>HostTransceiver::addSetMessage__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP 10.0.1.24 unsuccessful at attempt num  3 with id:  ID32 = 0x01000211 -> IND = 2, TAG = eoprot_tag_mc_joint_inputs_externallymeasuredtorque 
<#STRING_END#>
20222
48,881969
4
<#STRING_START#>HostTransceiver::addSetMessage__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  15 0 656 
<#STRING_END#>
20223
48,881975
4
<#STRING_START#>HostTransceiver::addSetMessage__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP 10.0.1.24 unsuccessful at attempt num  4 with id:  ID32 = 0x01000211 -> IND = 2, TAG = eoprot_tag_mc_joint_inputs_externallymeasuredtorque 
<#STRING_END#>
20225
48,881985
4
<#STRING_START#>HostTransceiver::addSetMessage__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  15 0 656 
<#STRING_END#>
20228
48,882001
4
<#STRING_START#>HostTransceiver::addSetMessage__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP 10.0.1.24 unsuccessful at attempt num  5 with id:  ID32 = 0x01000211 -> IND = 2, TAG = eoprot_tag_mc_joint_inputs_externallymeasuredtorque 
<#STRING_END#>
20230
48,882011
4
<#STRING_START#>HostTransceiver::addSetMessage__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  15 0 656 
<#STRING_END#>

```

 Attached to this issue there is the entire logger session.

[yarprunlog_26_05_2017_13_31_19.txt](https://github.com/robotology/icub-support/files/1031668/yarprunlog_26_05_2017_13_31_19.txt)

marcoaccame, 2017-06-05:
The origin of the problem is a communication CRC error in the left arm. These CRC errors cause packet losses which eventually give misbehavior in all robot parts beyond the left arm: right arm, torso, both legs. 

It is a very serious problem. Other issues can be caused by this CRC problem.

I recommend to check the cable connections in board 10.0.1.25 in ETH out port P3 and in ETH input of the next board. See also distance of the board from Electro-Magnetic noise.

```
from BOARD 10.0.1.25 (leftArm-j8_11), src LOCAL, adr 0, time 68s 813m 144u: 
(code 0x06000002, par16 0x0001 par64 0x0000000000000001) -> 
ETH monitor: detected RX CRC error in port specified by par16 (0 = P2, 1 = P3, 2 = internal). 
Number of errors in par64 + .
```

marcoaccame, 2017-06-05:
The problem in #468 could be due this.