robotology/mesh-support #475: Communication problem with BOARD EB7 IP 10.0.1.7

misaki43, 2017-06-12:
Hello,

### Description of the failure
each time we use the robot, we have a communication problem with the left leg. When I looked at the logs, I noticed other DEBUG information that seems to be abnormal.

### Detailed conditions and logs of the failure
 
The error information about the left leg are:

 ```
2560 869,380363 <ERROR> embObjMotionControl::getWholeImpedanceRaw() timed out the wait of reply from BOARD EB7 IP 10.0.1.7 joint  0 
 
2561 869,683254 <ERROR> embObjMotionControl::getWholeImpedanceRaw() timed out the wait of reply from BOARD EB7 IP 10.0.1.7 joint  1 
 
2562 870,097423 <ERROR> embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB7 IP 10.0.1.7 for nv ID32 = 0x0100000c -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_controlmodestatus 
 
2563 870,097481 <ERROR> An error occurred inside embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=pos) for BOARD EB7 IP 10.0.1.7
 
2564 870,097501 <ERROR> failure of embObjMotionControl::checkRemoteControlModeStatus(j=0, targetmode=pos) for BOARD EB7 IP 10.0.1.7 after 0 attempts and 0.301575 seconds
 
2565 870,097517 <ERROR> In embObjMotionControl::setControlModeRaw(j=0, mode=pos) for BOARD EB7 IP 10.0.1.7 has failed checkRemoteControlModeStatus()
 
2566 870,400326 <ERROR> embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB7 IP 10.0.1.7 for nv ID32 = 0x0100010c -> IND = 1, TAG = eoprot_tag_mc_joint_status_core_modes_controlmodestatus 
 
2567 870,400379 <ERROR> An error occurred inside embObjMotionControl::checkRemoteControlModeStatus(j=1, targetmode=pos) for BOARD EB7 IP 10.0.1.7
 
2568 870,400398 <ERROR> failure of embObjMotionControl::checkRemoteControlModeStatus(j=1, targetmode=pos) for BOARD EB7 IP 10.0.1.7 after 0 attempts and 0.303473 seconds
 
2569 870,400415 <ERROR> In embObjMotionControl::setControlModeRaw(j=1, mode=pos) for BOARD EB7 IP 10.0.1.7 has failed checkRemoteControlModeStatus()
```
The DEBUG information are:

 ```
228 1,412721 <DEBUG>  from BOARD 10.0.1.5/*1,3,6,7,8,9*/ (EB5), src LOCAL, adr 0, time 94s 997m 237u: (code 0x05000015, par16 0x0300 par64 0x0000000000103000) -> CFG: EOtheEncoderReader can be correctly configured. in p16 the number of joints (msb) and failure mask (lsb), in p64 for each physical joint a byte with (port, errorcode) + 
``` 
and:
 ```
609 13,177395 <DEBUG>      Device has no ICanBusErr interface
 ```
Here is a log file: 
[log_pc104_yarprobotinterface_3141.txt](https://github.com/robotology/icub-support/files/1067680/log_pc104_yarprobotinterface_3141.txt)

@serena-ivaldi @InriaBrice @dogoepp

julijenv, 2017-06-12:
Hi @misaki43 ,
This error message prevent you from putting the robot on its legs???? Or is it just an error message not relevant for the good behaviour of hte robot?

misaki43, 2017-06-12:
Hi! 
Currently we didn't try to put the robot on its leg but we hope we will do it soon.
So, currently we still can use the robot. If I understand correctly, you suggest we should ignore these errors, right?

julijenv, 2017-06-12:
Hi @misaki43 ,
No i didnt suggest this, actually but I wanted to know if you were using the legs nad if the legs were working even with this error pending.
As you are not using the legs, I would like you to tell me if the legs are enabled or not when yarprobotinterface has finsihed to calibrate the robot. In other words if the legs are ok or not after calibration ( view from yarpmotorgui).

julijenv, 2017-06-13:
Hi @misaki43 ,
Could you make some trials with the left leg and make some movements to understand what is the problem? Ethernet connection or encoder or motor. You can leave the robot on the pole for that no worries.

serena-ivaldi, 2017-06-14:
Hi @julijenv we cannot currently launch the robot - the yarpinterface fails to start; as it says:

```
EthResource::verifyBoardPresence() DID NOT have replies fro, board EB7 with IP 10.0.1.7 even after 20 attempts and 10.0156 seconds
EthResource::verifyEPprotocol() cannot verify BOARD EB7 with IP 10.0.1.7
```

julijenv, 2017-06-14:
Hi @serena-ivaldi ,
Can we receive the entire log?
please come on skype

marcoaccame, 2017-06-14:
Hi all,

1. does the board ping? on PC104 terminal digit: ping 10.0.1.7
2. if it does not ping: power motors off, then on, then ping it straight away.
3. when it pings, launch ethLoader and make sure the board has the correct application FW.
4. power motors off, then on. wait for 15 seconds after the boards begin to blink leds.
5. start yarprobotinterface. dont stop it even if there are problems on board 10.0.1.7.
6. send teh log.

@julijenv: if you need help on skype, pls call me.

marcoaccame, 2017-06-14:
```
EthResource::verifyBoardPresence() DID NOT have replies fro, board EB7 wit....
```
means that the board is either crashed, or in updater mode, or not reachable because of a broken eth cable. the previous actions are required to collect evidence on that.

julijenv, 2017-06-14:
Hi @misaki43 , @dogoepp , @serena-ivaldi .
Please come on skype for th

gabrielenava, 2017-06-14:
may be related to https://github.com/robotology/icub-support/issues/468

serena-ivaldi, 2017-06-14:
Hi @julijenv 

as anticipated on skype, we did as you said.

* Ping the board -> the board exist (and the others too)
* Check the boards firmware with ethLoader -> after clicking discovery they all appear, with the same firmware everywhere from the last update

Then we tried launching the robot twice:
* the first time, the robot launched, calibrated, but when we pressed CTRL+C to shutdown the robot it lost the communication with the board, giving many errors. The leg did not go to the resting position immediately, it was stucked while there were the communication errors (see log). When we pressed the second time CTRL+C, the motors were idled and the yarpinterface closed (even though with the classical errors and segmentation fault)

```
1516 164,041700 <ERROR> embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB7 IP 10.0.1.7 for nv ID32 = 0x0100000e -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone 
1517 164,041744 <ERROR> Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j=0) for BOARD EB7 IP 10.0.1.7
[Uploading 14_06_log_pc104_yarprobotinterface_ERROR_IN_SHUTDOWN.txt…]()
[14_06_log_pc104_yarprobotinterface_ERROR_NOT_EVEN_STARTING.txt](https://github.com/robotology/icub-support/files/1074444/14_06_log_pc104_yarprobotinterface_ERROR_NOT_EVEN_STARTING.txt)

```

* the second time, the robot startup failed. We launched the icubinterface, and the calibration didn't even started, because of these repeated communication errors as if the board was unreachable. But we could ping it. See again the log.

```
122 14,562813 <WARNING> EthResource::verifyBoardPresence() cannot reach BOARD EB7 with IP 10.0.1.7 at attempt # 20 w/ timeout of 0.5 seconds 
123 14,562865 <ERROR> EthResource::verifyBoardPresence() DID NOT have replies from BOARD EB7 with IP 10.0.1.7  even after  20  attempts and 10.0144 seconds: CANNOT PROCEED ANY FURTHER 
124 14,562890 <ERROR> EthResource::verifyEPprotocol() cannot verify BOARD EB7 with IP 10.0.1.7 : cannot proceed any further 
```

serena-ivaldi, 2017-06-14:
[14_06_log_pc104_yarprobotinterface_ERROR_IN_SHUTDOWN.txt](https://github.com/robotology/icub-support/files/1074450/14_06_log_pc104_yarprobotinterface_ERROR_IN_SHUTDOWN.txt)
[14_06_log_pc104_yarprobotinterface_ERROR_NOT_EVEN_STARTING.txt](https://github.com/robotology/icub-support/files/1074451/14_06_log_pc104_yarprobotinterface_ERROR_NOT_EVEN_STARTING.txt)

marcoaccame, 2017-06-14:
Hi @serena-ivaldi ... here are some answers:
1. you say `the first time, the robot launched, calibrated, but when we pressed CTRL+C to shutdown the robot it lost the communication with the board, giving many errors`: the behaviour is the same as in https://github.com/robotology/icub-support/issues/448. It was solved in devel branch with https://github.com/robotology/icub-main/commit/cd31d37e540f8d9d7720b2d8765db656e397e2f1.
2. you say `the second time, the robot startup failed. We launched the icubinterface, and the calibration didn't even started, because of these repeated communication errors as if the board was unreachable`: the behaviour is also described in the previous issue and was solved in devel branch with https://github.com/robotology/icub-main/issues/452.

If you dont want to wait for next imminent release of master branch, you can solve problem in point 2 by editing file iCubNancy01/hardware/electronics/pc104.xml file and set PC104RXrate = 2 or 1.

With this change yarprobotinterface will still crash w/ a CTRL-C (point 1 is not solved) but the next time it is restarted it will be able to reach all the boards.

An alternative to PC104RXrate = 2 or 1 is that you power off and on the motors.

marcoaccame, 2017-06-15:
Hi all,
I have compared the recent log files. There is not evidence of why the board 10.0.1.7 loses contact w/ PC104. I have however seen that the FW of the EMS is not aligned to master.

**I propose:**
1. align the system to master,
2. book a skype session w/ @julijenv to perform interactive tests.

## Analysys

In all logs, the board 10.0.1.7 stops communication. In the `log_pc104_yarprobotinterface_3141.txt` communication resumes even if after 90 seconds. 

There is not evidence of what is happening: no sign of ETH link up and down etc.

I have however seen that the FW version of the EMS boards is not aligned to master. It is v 2.52 and in master we have v 2.57.

```
446 7,496333 <INFO> EthResource::askBoardVersion() found BOARD EB7 @ IP 10.0.1.7 of type
ems4 with FW version = ( 2 , 52 ) and build date 2016 Sep 16 11:37 
```

```
SHA-1: 6585102f394a448ecbfca0c989886fdecf13b7d8

* new versions for ems (v 2.57), mc4plus (v 2.46) and mc2plus (v 1.14).

- built on 6 dec 2016 using icub-firmare w/ SHA-1: c6f788af203af9afb5762fac44301da85007cea8 
  (master tagged v1.6.0) and icub-firmware-shared w/ SHA-1: 
  e7000b876b01686c2c5644997d224c9ffad2f9ed (master tagged v1.6.0)
- main changes from previous release tagged v1.4.1 (16 sept 2016) are:
  - added hardware joint position limit management
  - max size of application increased to 384kb
  - applications made compliant to updater protocol version 2 (they can be queried by fw updater 
    tools without being forced jumping to updater soon after).
  - enhanced resolution of torque, stiffness, damping from milli to micro N (branch stiffDampFloat)
  - added support for reading the gyroscope which is mounted on EMS boards (branch emsgyro).
  - fixed bug in calib6 (branch bugfixcalib6)
  - heavy bugfix of motion control of forearm done by mc4plus boards
  - added support for configuring mc4 can boards to transmit motor positions up to pc104

* Tag Info:
object 6585102f394a448ecbfca0c989886fdecf13b7d8
type commit
tag v1.6.0
tagger marco.accame <marco.accame@iit.it> 1481033844 +0100

iCub Software release 1.6.0

```

dogoepp, 2017-06-29:
Hi, @julijenv found the issue : a broken magnetic encoder was causing board EB7 to malfunction. This discovery cust us an other such board.

The sensor has been replaced and EB10 (left leg skin) has been promoted to be EB7 (lower left leg) until we receive the new boards from IIT.

closed 2018-01-16 by julijenv