robotology/mesh-support #476: left pinky broken and right hand skin unavailable

amaroyo, 2017-06-12:
Right hand, thumb and index finger skin is not working. 
Left hand, pinky finger cable is broken, see picture below:

[image]

julijenv, 2017-06-12:
Hi @amaroyo ,
for the skin problem you may probably just need to open the dorso cover of the right hand to discover that the little pcb boards are a bit disconnected (not paralle to each other) to have all the fingertips aorking properly again.
for the palm you might consider taking a look at the wires going from the thumb base (where the wres from the palm gets out) up to the board above the MAIS.
Let me know

amaroyo, 2017-06-12:
Ciao @julijenv, if it was only looking at the boards to see if they were disconnected, I was not going to write you ¬¬ 

The MAIS has been working properly since the last time!! But the pinky finger tendom is broken, as you can see in the photo! We miss you here in Erzelli, that's why we randomly broke some finger :P :D

amaroyo, 2017-06-15:
fixed! :)

closed 2017-06-15 by amaroyo