robotology/mesh-support #477: [IcubBarcelona01]  Overheating of right wrist motor

jypuigbo, 2017-06-12:
The right arm wrist yaw motor (joint 6) overheats dangerously. It happens when running `ARE` in compliant mode and not in any of the following cases:
- manual control of the joint through `robotMotorGui`
- control of the joint through `ARE` in non-compliant mode

The overheat happens just after asking to home all joints in just a few seconds.

The left arm works nornaly and doesn't overheat under the same conditions, suggesting a hardware issue. 

How could we track down this issue?

@clement-moulin-frier @julijenv @sockchinglow

julijenv, 2017-06-12:
Hi @jypuigbo ,
COuld you tell me if the motor goes in some position in which it could be forced too much on 1 motor - to say it more clearly, could you check if the polso is correctly calibrated? Can you check if the polso can move to every of the extreme of the limit or not? From what I recall when operated ARE the polso are in some configurations and maybe the angular position for this specific polso cant go to the position and so force to try to reach the position and so start getting hot/warm.
let me know

jypuigbo, 2017-06-13:
The position was `home`. The joints were not forced. When we move them with `robotMotorGui` it can go to any range of it's movement without overheating. Moreover, the position is the same for both arms, and the left arm is not overheating in the same position.

julijenv, 2017-06-13:
Hi @jypuigbo ,
is there anyway that one of your teammate can come on skype and try to reach us:
skype Id: icub.support,
let me know

jypuigbo, 2017-06-13:
I'm connected to Skype and Alex is also with me.

jypuigbo, 2017-06-14:
Apparently it was that the joint was blocked at some point before reaching the joint safety limit, what was causing the overheating. 

After changing the limits of the joints and fine-tuning the arm calibration to make the new limits fall in a safe range, the joint doesn't overheat anymore. 

Thanks @julijenv for the help.

julijenv, 2017-06-15:
U r welcome guys, @jypuigbo  @Alex

closed 2017-06-15 by julijenv