robotology/mesh-support #480: Updating the laptop of iCubNancy01

serena-ivaldi, 2017-06-14:
Dear all

the IIT laptop of iCubNancy01 is currently in Ubuntu14 and we need to upgrade it to Ubuntu16.
Are there specific procedures to be followed? 
Or can we just make a dist-upgrade?

traversaro, 2017-06-14:
Perhaps @mbrunettini can point to the correct procedure to following in this cases.

mbrunettini, 2017-06-15:
A dist-upgrade should be fine,  but please check any configuration file in case of a new version may replace what we configured: during the upgrade, when system asks what to do in case of file overwritting, please select "compare" and check modifications.
After the upgrade, remember to re-enable the additional sources (specially the icub one).
Does this laptop work as server (i.e. exports the robot source code)?

serena-ivaldi, 2017-06-15:
Yes it exports the robot source code. As a backup, I was thinking about making an exports repo on our machine... but of course we need to change the nfs parameters on the iCub.

mbrunettini, 2017-06-16:
I think it is not needed if you commit and push you code, worst case you can simply copy you repo to another machine.
if you want to check afterwards, [here ](http://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions)you can find a guide about setting up the server laptop.
let us know, I keep this opened until upgraded is finished.

closed 2018-01-16 by julijenv