robotology/mesh-support #482: Returning to home position goes wrong

clement-moulin-frier, 2017-06-20:
Hi,

### Description of the failure

Sometimes `iCubBarcelona01` doesn't return well in the home position, as shown in the picture below. This position is after a `calibration stop` in `/iolReachingCalibration/rpc`. I also happens when sending `home all` to `/actionsRenderingEngine/cmd:io` (see detailed description below):

[image]

### Detailed conditions and logs of the failure

This doesn't happen all the time but sufficiently often to prevent working correctly with the robot. First time we observed it was when doing `iolReachingCalibration` (after 3 reaching movements). It also sometimes happens when a module send `home all` to `ARE`. Once it happens, `home all` is not able to put the robot back in the home position anymore, it just moves a little bit, very slightly. However, if we set the shoulder motors in `position` mode and make them move a bit with `yarpmotorgui`, then `home all` works as expected.

For info, we had a wrist issue a few days ago #477 . We solved it but today we observed that the left wrist was heating as well.

Thank for the help,
Clément

Cc @sockchinglow @jypuigbo @julijenv

julijenv, 2017-06-22:
Hi @clement-moulin-frier ,
could you please give us the enocder values for the entire 16 joints in orde rfor us to try to understand.
thx in advance

clement-moulin-frier, 2017-06-22:
Hi @julijenv ,

Yes sure, here is more info. We just tried again doing `iolReachingCalibration`. This time, calibration with the left arm went fine (4 reaching movements). The problem appeared after the fourth reaching movement on the right arm, where the robot went in this position after the `calibration_stop` command (so this time the problem was with the other arm that in my previous post):

[image]

(Note that we also observed the problem outside of a `iolReachingCalibration`, when some module call `home all` in `ARE`, as described in my previous post.)

Here are the motor encoder values of the [left arm](https://user-images.githubusercontent.com/6000622/27445103-eeca19ba-5778-11e7-89ec-dfdbc91a2fbb.png) and the [right arm](https://user-images.githubusercontent.com/6000622/27445158-1322b8d0-5779-11e7-9226-4a904050249d.png) (sorry they are just screenshots, we didn't find a way to save those values from yarpmotorgui. You can still see the greyed value by zooming in)

More info in case it can help:

We observed that sometimes we start `ARE` and the port `/actionsRenderingEngine/cmd:io` is not created. If we stop it and start it again, then we have access to the port. This happened today, here are the [ `ARE` logs when we run it the first time (port not created)](https://github.com/robotology/icub-support/files/1095509/log_icubAlien_actionsRenderingEngine_23361__no_rpc.txt). And here are the
[`ARE` logs after having restarted it and observed the problem described above](https://github.com/robotology/icub-support/files/1095510/log_icubAlien_actionsRenderingEngine_29811_wrongpos.txt).

Here are [the log of `iolReachingCalibration`](https://github.com/robotology/icub-support/files/1095512/log_icububuntu_iolReachingCalibration_3623.txt)

A last point: today we observed a heating in the left wrist (see #477 ). So we'll follow the same procedure on the left side. It's weird however that both wrists got the same problem within a week.

Hope this helps.

Thank you,
Clément

clement-moulin-frier, 2017-06-27:
Hi,

Any news on this issue? Please let us know if the screenshots we sent regarding the encoder values are sufficient or if we should better copy/paste the values from the yarpmotorgui interface.

Best,
Clément

pattacini, 2017-06-27:
Issue moved to [robotology/icub-main #455](https://github.com/robotology/icub-main/issues/455) via [**ZenHub**](https://www.zenhub.com/)

closed 2017-06-27 by pattacini