robotology/mesh-support #483: yarprobotinterface error: cannot find parameter fullscalePWM

gsaponaro, 2017-06-21:
Hi, we updated all software repositories to the latest **master branch**: robots-configuration, yarp, icub-main, icub-firmware-shared.

yarprobotinterface does not start anymore on the iCubLisboa01 pc104 because it does not find the parameter `fullscalePWM`.

Full log here:
https://gist.github.com/gsaponaro/248aee5b8dba7a73efbb38dd4f4fbdbd

I guess that we need to add the parameters `fullscalePWM` and `ampsToSensor` to our robot configuration xml files.

To do that, should I simply copy the parameter values from here?
https://github.com/robotology/robots-configuration/pull/25
https://github.com/robotology/icub-support/issues/451
And do you suggest that I make a PR in robots-configuration master or devel branch?

randaz81, 2017-06-27:
Hi @gsaponaro, the issue should be fixed now. Please tell me if everything is ok. Thank you

gsaponaro, 2017-06-27:
Yep, it works, thanks!

closed 2017-06-27 by gsaponaro