robotology/mesh-support #496: Upgrade cmake on pc104

gabrielenava, 2017-06-27:
we tried to compile yarp on pc104, after updating the source code to the newest version (master). Apparently, cmake 3 is required:

```
 CMake Error at CMakeLists.txt:5 (cmake_minimum_required):
 CMake 3.0 or higher is required.  You are running version 2.8.9

```
how can we update cmake? sudo-apt-get is enough or something else is required?
This is our pc104 version:

```
iCub LIVE image for PC104 version 5.0-16.02.12
Based on wheezy amd64
Build on 12-Feb-2016
Linux pc104 3.2.0-4+icub.1-rt-amd64 #1 SMP PREEMPT RT Debian 3.2.57-3 x86_64
```
Or should we move to a new image with ubuntu LIVE?

cc @barbalberto @mbrunettini

francesco-romano, 2017-06-27:
Hi gabri.

I am afraid that the maximum cmake supported on wheezy is 2.8.11.
you will have to update the pc104.

To be sure, wait confirmation from @mbrunettini

traversaro, 2017-06-28:
Unfortunately this was not reminded in the recent release notes of YARP 2.3.70 and ICUB 1.8.0, but since the latest release of YARP and ICUB we do not support anymore Wheezy, see https://github.com/robotology/QA/issues/173 . 

This means that we never tested those release on Wheezy and if it was necessary to require a newer version of a given library we used it, even if it was not available in Wheezy (see http://wiki.icub.org/wiki/YARP_Supported_Distributions, this is for example the reason why we require CMake 3.0). In particular, the C++11 support in the GCC version available in Wheezy is not complete, so it is also possible that some portions of the new C++11 code in YARP/ICUB do not compile in Wheezy. It is also possible to have ABI problems when linking against libraries with a C++ interface distributed with the system, as in GCC < 5 the ABI for C++11 and C++98 STL library is different. 

**In short, it is highly recommended that all users that want to use YARP version 2.3.70 or later on their iCub robot should update the OS on pc104 to the 6.0 images based on Debian Jessie, following the instructions in http://wiki.icub.org/wiki/The_Linux_on_the_pc104 .**

barbalberto, 2017-06-29:
Yes, please update the PC104 image. NOTE the images are Debian based, not ubuntu

closed 2018-04-09 by julijenv