robotology/mesh-support #498: Jaw Broken

arrenglover, 2017-06-28:
The plastic jaw has snapped. During neck pitching upwards while the head was yawed to the left (during the ball demo) the jaw got caught on the motors of the neck causing it to be broken.

To prevent it happening again, the jaw needs to be mechanically fastened closed, or the motors that control the jaw need to be operational (and exposed to the yarpmotorgui).

marcoaccame, 2017-06-28:
I have just had a look at the status of the face. 

There are mechanical fixes to be done:
1. jaw is broken (i understand that @spada1978 is going to replace it using the jaw of icub-v3),
2. eyelid movement is not mounted,
3. joints in left and right lips miss some metal pieces,
4. motor of upper lip which is attached to the skull is missing.  

After the mechanical fixes are done I shall test the xml files and we can have a moving operational face.
If a complete mechanical fix takes too long, we can move only a subset of the face. For instance the jaw only. Please @arrenglover tell me which joints of the face can be excluded.

arrenglover, 2018-01-24:
the face and jaw motors were installed and can be controlled to prevent this happening again

closed 2018-01-24 by arrenglover