robotology/mesh-support #499: FT sensors sometimes silently send NaN as measured force on right foot

gabrielenava, 2017-07-01:
We observed that sometimes, the FT sensors retrieve NaN as measured force on right foot. 

While the yarpRobotInterface and wholeBodyDynamics were running (and after FT calibration with `calib all`), we opened the port for reading the external forces acting on the right foot. The values were all NaN. The values for the left foot instead were ok.

No errors, warnings or other messages are printed in the logger. A restart of  the yarpRobotInterface solved the issue every time it happened (it happened 3 times so far)

traversaro, 2017-07-01:
> we opened the port for reading the external forces acting on the right foot. The values were all NaN

If I understood correctly, the NaN were on the port published by `wholebodydynamics`. Why do you infer that the NaN were published by the F/T sensor?

gabrielenava, 2017-07-02:
My mistake, I didn't verify the FT directly. Maybe the next time it is worth to verify if the FT measurements are sound, and eventually move this issue in another location

gabrielenava, 2017-07-02:
cc @dogoepp @serena-ivaldi

diegoferigo, 2017-09-27:
We noticed the same problem also on `iCubGenova04`. Today it happened at least two times. In this case, the nan was in the external forces on the right arm and the torso, both times.

traversaro, 2017-11-29:
@fjandrad do you remember anything about this issue?

traversaro, 2017-11-29:
Now I remember. This was a bug in the `wholebodydynamics` device, fixed by https://github.com/robotology/codyco-modules/pull/243 . The iDynTree API was modified in https://github.com/robotology/idyntree/pull/378 to avoid this kind of pitfalls in the future. Closing.

closed 2017-11-29 by traversaro