robotology/mesh-support #500: Meta-issue to track robot-specific configuration problems in the neck calibration

traversaro, 2017-07-01:
See https://github.com/robotology/robots-configuration/issues/36 (filled in `robots-configuration` as suggested in https://github.com/robotology/robots-configuration/issues/17).

This is just a meta-issue to make sure that the iCub support team knows that a set of robots is affected by this issue (using the robot-specific GitHub labels).

traversaro, 2017-07-01:
cc @davidetome @julijenv @gabrielenava

davidetome, 2017-07-03:
it happens on CAN based robots, you have to play with

valegagge, 2017-08-22:
All these robots have CAN heads, but in their calibration head files is configured ParametricCalibrationEth instead of ParametricCalibration.

The ParametricCalibration doesn't use calibrationZero and CalibrationDelta parameters, so the user need to modify Zero parameter in the motorcontrol file as @davidetome had already wrote.

So to fix this issue we should change the calibrator device type and remove those parameters in order to not bring the user on wrong way.

For the future, could we update ParametricCalibration device in order to manage  calibrationZero and CalibrationDelta?

I don't know if it is possible and how much effort is needed.

closed 2019-10-12 by pattacini