robotology/mesh-support #504: Complete update of iCubHeidelberg01

hu-yue, 2017-07-04:
Hi,
 It has been a while since we have updated anything of our iCub (I think last time was in November 2016).
Since last month there has been a release of many things, I am planning a major update for our iCub next week, after an event where I would still need the robot up running with the current demos.

I would like to have some feedback on what is better/what should be done and use this issue to keep track of possible problems we might have during the upgrade.

According to the issue https://github.com/robotology/QA/issues/240, I have the following questions:

* Currently on the server we are running Ubuntu 14.04, should it be updated to 16.04? (I guess sooner or later this will have to be done)
* For the firmware, in issue https://github.com/robotology/QA/issues/221 it was suggested to perform the FW upgrade with icub-support, shall we schedule a date after I have performed the steps before (OS updagrade etc)? @marcoaccame 

Thank you!

francesco-romano, 2017-07-04:
Hi @hu-yue 
For the operating system: yes. You should upgrade everything (also the pc104) to 16.04 as YARP now requires C++11 and CMake 3.0 (as AFAIK they are not included in 14.04)

(@traversaro for confirmation)

traversaro, 2017-07-04:
I think that the pc104 needs to be updated to the latest image based on Debian Jessie, as documented in https://github.com/robotology/QA/issues/240 .

hu-yue, 2017-07-04:
@francesco-romano Ok perfect, thanks!
@traversaro yes I saw that thanks! There will be a lot to be upgraded =)

marcoaccame, 2017-07-05:
Hi @hu-yue, 
I recommend to do first a full system update as described in https://github.com/robotology/QA/issues/240 with eApplication firmware only. The updating of eLoader / eUpdater is a plus, not a requirement.

Then, check if you really need to upgrade from old versions of eLoader / eUpdater.
- with FirmwareUpdater program send all the ETH boards in maintenance, as described in section (2).a of document https://github.com/robotology/icub-firmware-build/blob/master/ETH/FirmwareUpdater.readme.maintenance.of.eth.robots.txt.
- very simply, if a board is in maintenance mode and you can read the kind of `Board` and edit the `Info` field you already have recent versions of eLoader / eUpdater.
- otherwise, you have old versions of eLoader / eUpdater and you can do the update. 

Since my previous post https://github.com/robotology/QA/issues/221, I have published detailed instructions of how to perform the FW update of eLoader and eUpdater. 

Hence, in case you want to perform the update by yourself follow sections (5) and (6) of https://github.com/robotology/icub-firmware-build/blob/master/ETH/FirmwareUpdater.readme.maintenance.of.eth.robots.txt.

If instead you prefer to be helped in the operation, please arrange a session with @julijenv. I will join him and together we shall guide you through.

hu-yue, 2017-07-10:
Hi @marcoaccame sorry for the late reply, thank you for the instructions! I will try to go through everything myself and contact you in case of problems.

hu-yue, 2018-01-10:
Hi, 
after another long time, I finally restarted the updates.
Today I tried to update the firmware of the robot, after having updated already all the repositories and recompiled yarp and icub.
However, I am not sure about the outcome. I started with the ETH firmware of boards 10 and 11 and the board status says `application (IDLE)` instead of just `application` as in the boards that I have not yet updated. So I updated also the `eLoader` and `eUpdater`, and then reuploaded the FW but it still says this. Since I am not sure that `IDLE` should be there, I would like a feedback before keeping going on with all the rest of the boards.

Screenshot of the FirwareUpdater:

[image]

Thanks!

hu-yue, 2018-01-10:
Not sure if this is anyway meaningful or useful, but when uploading the FW with `Upload Application` in the FirmwareUpdater, I get these outputs in the terminal:
```
EthMaintainer::program() is about to program the board @ 0.0.0.0 of type none with an eApplication.
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.5: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.6: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.10: ems4 w/ eUpdater v 2.23 running protocol v 2 w/ capabilities = 0x607ffbfa. mainteinance = ON
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.7: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.8: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.11: ems4 w/ eApplication v 3.5 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.9: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.5: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.6: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.10: ems4 w/ eUpdater v 2.23 running protocol v 2 w/ capabilities = 0x607ffbfa. mainteinance = ON
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.7: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.8: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.11: ems4 w/ eApplication v 3.5 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF
EthMaintainer::processDiscoveryReplies2() has found board @ 10.0.1.9: ems4 w/ eApplication v 2.55 running protocol v 2 w/ capabilities = 0x60030030. mainteinance = OFF

WARNING: EthMaintainer::program() has found 1 boards @ 0.0.0.0.
- board 1 of 1: 10.0.1.10
EthMaintainer::cmdProgram() is about to program the APP partition of 1 boards: 10.0.1.10 (candoit)
OK: EthMaintainer::program() has succesfully programmed board none @ 0.0.0.0.

```

@marcoaccame 
Thanks!

hu-yue, 2018-01-10:
Ok so I just updated also board 5 (torso) and started the robot with torso only and it seems that there are not special complains, should I interpret this as `application (IDLE)` not being an issue?

hu-yue, 2018-01-10:
So I updated all the firmware (all boards are showing `application (IDLE)`, and everything runs fine, except the IMU, and I am not sure why, please see error log below:

```
||| finding paths [plugins]
Failed to open com port /dev/ttyS0
yarpdev: ***ERROR*** driver <xsensmtx> was found but could not open
[ERROR]cannot create device <xsensmtx>
[ERROR]ControlBoardWrapper: error while opening subdevice
yarpdev: ***ERROR*** driver <inertial> was found but could not open
[WARNING]Cannot open device inertial 
[WARNING]Cannot open device inertial 
||| finding paths [plugins]
[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[WARNING]There was some problem opening one or more devices. Please check the log and your configuration 
[ERROR]One or more devices failed opening... see previous log messages for more info 
[ERROR]Error in startup phase... see previous messages for more info 
[WARNING]Interrupt # 1 # received. 
[INFO]Interrupt received. Stopping all running threads. 
[INFO]interrupt1 phase starting... 
[INFO]Entering action level 1 of phase interrupt1 
[ERROR]park device do not exists 
[ERROR]Cannot run park action on device left_leg-calibrator 
[ERROR]park device do not exists 
[ERROR]Cannot run park action on device right_leg-calibrator 
[ERROR]park device do not exists 
[ERROR]Cannot run park action on device torso-calibrator 
[INFO]All actions for action level 1 of interrupt1 phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 1 of interrupt1 phase finished. 
[WARNING]There was some problem running actions for interrupt1 phase . Please check the log and your configuration 
[INFO]interrupt1 phase finished. 
[ERROR]Error in interrupt1 phase... see previous messages for more info 
[INFO]shutdown phase starting... 
[INFO]Entering action level 5 of phase shutdown 
[ERROR]torso-mc_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device torso-mc_wrapper 
[ERROR]right_leg-FT_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device right_leg-FT_wrapper 
[ERROR]left_leg-FT_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device left_leg-FT_wrapper 
[ERROR]right_foot-FT_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device right_foot-FT_wrapper 
[ERROR]left_foot-FT_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device left_foot-FT_wrapper 
[ERROR]right_leg-VFTserver is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device right_leg-VFTserver 
[ERROR]left_leg_VFTserver is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device left_leg_VFTserver 
[ERROR]torso-VFTserver is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device torso-VFTserver 
[ERROR]left_leg-mc_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device left_leg-mc_wrapper 
[ERROR]right_leg-mc_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device right_leg-mc_wrapper 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
[WARNING]There was some problem running actions for shutdown phase . Please check the log and your configuration 
[INFO]shutdown phase finished. 
[ERROR]Error in shutdown phase... see previous messages for more info 
[INFO]RFModule failed to open.

```

julijenv, 2018-01-11:
Hi @hu-yue ,
I've got asking to @marcoaccame , and it tells me that you should keep going updating the application on every EMS boards, the (IDLE) message means everything OK.
Let us know

hu-yue, 2018-01-11:
Hi @julijenv thanks for your feedback!

In the meantime, I updated all the FW, and the robot can run everything and everything seem to work, except for the IMU, basically it is seems fine if I exclude `<devices file="hardware/inertials/head-inertial.xml" /> ` from the configuration.
Otherwise `yarprobotinterface` would give the errors listed in the previous comment and would not start:
```
Failed to open com port /dev/ttyS0
yarpdev: ***ERROR*** driver <xsensmtx> was found but could not open
[ERROR]cannot create device <xsensmtx>
[ERROR]ControlBoardWrapper: error while opening subdevice
yarpdev: ***ERROR*** driver <inertial> was found but could not open
[WARNING]Cannot open device inertial 
```
So far I could not figure out a solution. 

Apart from this, I also noticed that the new Live OS on the pc104 seems very slow (takes about one and half minute to start, before it was definitely faster), and for some reasons it never boots when I switch it on once, I always had to switch it off and then on again.

hu-yue, 2018-01-15:
Hi, @robotology/icub-support-team, I would like to ask if you could give me some feedback on the error on the `xsensmtx` as in the previous comment? Because in the current situation we are unable to use the robot for most of our modules. Thanks!!

julijenv, 2018-01-15:
Hi @hu-yue ,
From the log file you sent us, I cant see the error related to <xsensmtx> or at least the same problem as yours, withthe /dev/ttyS0 not opened.
Could you please doucle check that the xsens was mounted correctly by looking into:
dmesg | grep TTYS
and see if the xsens appears.
R u sure that u tick the arrow for the compilation of the xsens on pc104.
LEt us now.
Please resend the log file with the erros mentionned in your previous mail.
thx in advance

hu-yue, 2018-01-15:
Hi @julijenv thanks for the reply.
I checked `dmesg | grep TTY` and nothing appeared, does this mean that the xsens is not mounted correctly then?
I checked also the compilation of `icub-main` on pc104, and `xsensmtx` is set on `ON`, please see the CMakeCache: [CMakeCache.txt](https://github.com/robotology/icub-support/files/1632325/CMakeCache.txt)

I attach the log of `yarprobotinferface`, not from `yarplogger` because it seems it's skipping some lines, this time there should be the error mentioned (starting line 302): [log.txt](https://github.com/robotology/icub-support/files/1632340/log.txt)

Thanks!

hu-yue, 2018-01-15:
@julijenv sorry my mistake, with `dmesg | grep tty` I get the following outputs:
```
[    0.000000] console [tty0] enabled
[    0.974838] 00:02: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    0.995402] 00:03: ttyS1 at I/O 0x2f8 (irq = 3, base_baud = 115200) is a 16550A
```

julijenv, 2018-01-16:
Hi @hu-yue ,
could you try to get some insides of who is what on those 2 tty, hopefully just the number of the xsens has been swapped from 0 to 1, or worse 0 or 1 or neither xsens. try to read the entire dmesg and check more details dispalyed around of the detection of tty driver launched.

hu-yue, 2018-01-16:
Hi @julijenv, I solved my problem.
The problem was that the user `icub` was not in the `dialout` group, therefore I got the message `Failed to open com port /dev/ttyS0`. I added the `icub` user to the `dialout` group and now it works.
I leave the issue open for now as I might get other issues by proceeding further now that the IMU works again.
Thanks again.

hu-yue, 2018-01-16:
Last issue I had is the estimation of `wholebodydynamics`, in the `iCubHeidelberg01/estimators/wholebodydynamics.xml` file currently on `codyco-modules`, there is the following list of devices:
```
        <action phase="startup" level="15" type="attach">
            <paramlist name="networks">
                <!-- motorcontrol and virtual torque sensors -->
                <elem name="left_lower_leg">left_lower_leg_mc</elem>
                <elem name="right_lower_leg">right_lower_leg_mc</elem>
                <elem name="left_upper_leg">left_upper_leg_mc</elem>
                <elem name="right_upper_leg">right_upper_leg_mc</elem>
                <elem name="torso">torso_mc</elem>
                <!-- imu -->
                <elem name="imu">inertial</elem>
                <!-- ft -->
                <elem name="l_leg_ft_sensor">left_upper_leg_strain</elem>
                <elem name="r_leg_ft_sensor">right_upper_leg_strain</elem>
                <elem name="l_foot_ft_sensor">left_lower_leg_strain</elem>
                <elem name="r_foot_ft_sensor">right_lower_leg_strain</elem>
            </paramlist>
        </action>
```
which gives the error `[ERROR]Target device left_lower_leg_mc (network = left_lower_leg ) does not exist.`, because the current names of the devices are different as in `robots-configurations`. So I updated the file as follows:
```
        <action phase="startup" level="15" type="attach">
            <paramlist name="networks">
                <!-- motorcontrol and virtual torque sensors -->
                <elem name="left_lower_leg">left_leg-eb7-j4_5-mc</elem>
                <elem name="right_lower_leg">right_leg-eb9-j4_5-mc</elem>
                <elem name="left_upper_leg">left_leg-eb6-j0_3-mc</elem>
                <elem name="right_upper_leg">right_leg-eb8-j0_3-mc</elem>
                <elem name="torso">torso-eb5-j0_2-mc</elem>
                <!-- imu -->
                <elem name="imu">inertial</elem>
                <!-- ft -->
                <elem name="l_leg_ft_sensor">left_leg-eb6-j0_3-strain</elem>
                <elem name="r_leg_ft_sensor">right_leg-eb8-j0_3-strain</elem>
                <elem name="l_foot_ft_sensor">left_leg-eb7-j4_5-strain</elem>
                <elem name="r_foot_ft_sensor">right_leg-eb9-j4_5-strain</elem>
            </paramlist>
        </action>
```
and now everything seems to work.
@traversaro fyi.

I am again getting the `[ERROR]hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.8. Expected: 156630, Received: 156631, Missing: 1, Prev Frame TX at 820705990 us, This Frame TX at 820715993 us` error, now on boards `5` and `8`, as in #83 , however I had the impression that it did not appear before, when I didn't include the `IMU` and the `wholebodydynamics` estimation, because the errors appeared right after the calibration of the `FT` sensors ended at startup, they don't appear again if I do another calibration via `rpc` though.

traversaro, 2019-10-03:
As the update was completely successful y, I think we can close the issue. If the error messages reported in https://github.com/robotology/icub-tech-support/issues/504#issuecomment-357973174 are still a problem, a new issue can be opened.

closed 2019-10-03 by traversaro