robotology/mesh-support #505: Motor control malfunction

alelais, 2017-07-05:
Hello,
the motor control of our iCub seems not to work properly. We want to find out, whether it is a hardware or a software issue. In case of a hardware issue we want to track down which components are broken in particular. Our current assumption is, that something is wrong with the power supply of the motors.

The motors (except the eye lids) of the robot remain loose after the boot up, regardless whether the fault button is pressed or released.

Our power supply looks like this when the robot is turned off:
[image]

and like that, when the robot booted up:
[image]

As we want to find out, whether the motor control block of our iCub is malfunctioning, it is maybe useful to take a look at the involved boards. Here is the main power distribution card:

julijenv, 2017-07-10:
Hi @alelais ,
could you please add a log file of the yarprobotinterface,
that could help us a lot to debug.
thx in advance

alelais, 2017-07-10:
Hi,
I couldn't launch yarprobotinterface properly, as there were no `yarprobotinterface.ini` and a couple of `*.xml` files neither in `/home/icub/.local/share/yarp/robots/iCubFrankfurt01/` nor in `/usr/local/src/robot/robots-configuration/iCubFrankfurt01/`. Therefore, I copied those from another group and modified them for the use with our robot. I had followed the robot configuration manual at `http://wiki.icub.org` back then when I was setting up the robot, though the mentioned files were not present. What is the proper way of generating those configuration files? Could you maybe show me a reference?

After I have included the mentioned files, the output of yarprobotinterface looked like this:
```
icub@pc104:~$ yarprobotinterface
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/.local/share/yarp/robots/iCubFrankfurt01/yarprobotinterface.ini] (pwd)
||| found /home/icub/.local/share/yarp/robots/iCubFrankfurt01/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/.local/share/yarp/robots/iCubFrankfurt01/./icub_head.xml] (pwd)
||| found /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./icub_head.xml
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./icub_head.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./icub_head.xml at line 2 . "robot" element should contain the "build" attribute [unsigned int]. Assuming 0 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./icub_head.xml at line 2 . "robot" element should contain the "portprefix" attribute. Using "name" attribute 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./wrappers/motorControl/head_mc_wrapper.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./wrappers/motorControl/head_mc_wrapper.xml at line 4 . Trying to import a file for the wrong robot. Found iCubFrankfurt01 instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./wrappers/motorControl/head_mc_wrapper.xml at line 4 . Import a file for a different robot build. Found 1 instead of 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./hardware/motorControl/icub_head.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./hardware/motorControl/icub_head.xml at line 3 . Trying to import a file for the wrong robot. Found iCubFrankfurt01 instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./hardware/motorControl/icub_head.xml at line 3 . Import a file for a different robot build. Found 1 instead of 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./general.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./general.xml at line 4 . Trying to import a file for the wrong robot. Found iCubFrankfurt01 instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./general.xml at line 4 . Import a file for a different robot build. Found 1 instead of 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./hardware/inertial.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./hardware/inertial.xml at line 2 . Trying to import a file for the wrong robot. Found iCubFrankfurt01 instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./hardware/inertial.xml at line 2 . Import a file for a different robot build. Found 1 instead of 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./calibrators/head_calib.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./calibrators/head_calib.xml at line 3 . Trying to import a file for the wrong robot. Found iCubFrankfurt01 instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./calibrators/head_calib.xml at line 3 . Import a file for a different robot build. Found 1 instead of 0 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./general.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./general.xml at line 4 . Trying to import a file for the wrong robot. Found iCubFrankfurt01 instead of icub 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubFrankfurt01/./general.xml at line 4 . Import a file for a different robot build. Found 1 instead of 0 
yarp: Port /icub/yarprobotinterface active at tcp://10.83.255.18:10004/
[INFO]startup phase starting... 
[INFO]/icub/head : no ROS initialization required 
[INFO]/icub/head  initting YARP initialization 
yarp: Port /icub/head/rpc:i active at tcp://10.83.255.18:10005/
yarp: Port /icub/head/command:i active at tcp://10.83.255.18:10006/
yarp: Port /icub/head/state:o active at tcp://10.83.255.18:10007/
yarp: Port /icub/head/stateExt:o active at tcp://10.83.255.18:10008/
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
||| finding paths [plugins]
||| checking [/home/icub/.local/share/yarp/robots/iCubFrankfurt01/plugins] (pwd)
[... cut ...]
[ERROR]head_mc_wrapper is not a wrapper, therefore it cannot have detach actions 
[ERROR]Cannot run detach action on device head_mc_wrapper 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
[WARNING]There was some problem running actions for shutdown phase . Please check the log and your configuration 
[INFO]shutdown phase finished. 
[ERROR]Error in shutdown phase... see previous messages for more info 
[INFO]RFModule failed to open.
```

julijenv, 2017-07-11:
Hi @alelais ,
Unfortunately I am not in charge of this generation of files. I wanted you to launch yarprobotinterface because I thought you had the last updates but if you didnt it isnt a problem. With what were you launchinfg the head? robotInterface? Do as you usually do and send me back the logfile.
From the logfile you sent to me it appears that the cfw2can is not reachable, I hope it is just becasue you tried to launch YARProbotinterface.
Let me know.
If ever needed Im behind my dedssktop, ring me on skype, my ID is icub.support

julijenv, 2017-08-04:
Hi @alelais ,
If just see in quite a late matter your reply on skype.
Justs to make sure the power needed by the head can sometimes be more than expected so it is important to give at least 5A one head adn motors side ( for the CPU 13,1V is more a usual value)
After starting the head , try to launch (with motor ON) the application
canLoader (net 0, cfw2can AND connect)
and check if you can see the boards.
Please next time that u try to get in touch with me on skype, please ring me

julijenv, 2017-11-13:
Hi @alelais ,
need the password of your laptop to start debugging the issue.
Please send it to me. I already sent you a mail on your mail add .

closed 2018-04-09 by julijenv