robotology/mesh-support #509: Right Arm Shoulder (and maybe also wrist) Broken

giuliavezzani, 2017-07-12:
### Description of the failure
Tendons of right shoulder broken.
To check if right wrist is broken or not.

claudiofantacci, 2017-07-12:
Shoulder tendons 2 and 4 broken, 7 needs to be checked.

Wrist seems to be ok, further investigation may be needed.

One of the right thumb tendon seems to be loose. It needs further investigations.

julijenv, 2017-07-13:
Hi @giuliavezzani  and @claudiofantacci ,
it has been changed shoulder cables 2 and 4, the cable 7 has been remounted without being changed, maybe later Ill change it.
The thumb was looking alright strangely enough when I asse this morning (weird though).
We''l see.

IITCODE Qta Description
4759 x 1 : shoulder rotation cable 2
4761 x 1 : shoulder elevation cable 4
1896 x 2 : right pulley crimp cover

closed 2017-07-13 by julijenv