robotology/mesh-support #510: Communication problem between boards - left arm not responding

pi-q, 2017-07-13:
It seems that one of the boards on the left arm is not responding.

The logs showing the failure are attached.
[log_icub-head_yarprobotinterface_4218.txt](https://github.com/robotology/icub-support/files/1144737/log_icub-head_yarprobotinterface_4218.txt)

marcoaccame, 2017-07-20:
The log shows that the ETH link between boards EB25 and EB3 is not healthy. At first there are detected CRC errors, then the link keeps on going down and up again. Hence there are packet loss as the log say.

Hence, I recommend to view connections between the two boards and also possible current leaks close to board EB3.

julijenv, 2017-08-22:
during my debug on another issue #515 , i had to recheck the overall wiring and I probably redo this bad connection, because when restarted lately it doesnt show up anymore this error.
fingers crossed

closed 2017-08-22 by julijenv