robotology/mesh-support #511: iCub not moving when running yarprobotinterface

clement-moulin-frier, 2017-07-18:
Cc @julijenv @pattacini @jypuigbo @sockchinglow @Tobias-Fischer 

### Description of the failure

When starting `yarprobotinterface`, the robot doesn't move. When looking at the logs, there are a lot of timout errors like this:
`100ms Timeout occured in setControlModeRaw ( j:1), current mode: idl, requested: pos`.

Here is [the full log](https://github.com/robotology/icub-support/files/1156392/log_pc104_yarprobotinterface_1004.txt).

### Detailed conditions and logs of the failure

Since a couple of weeks we had problems with the ikart computer not starting time to time. Yesterday it was not starting at all anymore, so we opened it with Alex. We first thought it was due to the power supply we previously installed for the ikart computer. We changed it, still same problem. Then we saw that the graphic card fan was not moving correctly. We changed the graphic card and the ikart computer started correctly. End of the day.
Then today, we tried to start the iCub. When running `yarprobotinterface`, the robot doesn't move. Looking at the logs, there are a lot of timeout errors saying motors are in idle instead of pos (see above)

When running `canLoader` on the `pc104` we get this:
[image]

Thanks,
Clément

Tobias-Fischer, 2017-07-18:
In case it's related to the torso boards, it seems like the lights of the boards are not switched on. Here is a photo:
https://wysiwyd.slack.com/files/clement_mf/F69NMR29F/img_20170718_165525.jpg
I am not in Barcelona though, so I might be totally wrong .. in that case, please ignore ;)

julijenv, 2017-07-19:
HI @clement-moulin-frier ,
the robot is not moving at all?!?!?
did you press the red button by any chance?!?!
I red the log and it's hard to understand what's going wrong so the best will be to get in touch.
I'm available on skype, i will wait for you.

julijenv, 2017-08-14:
Hi @clement-moulin-frier ,
we cleared this issue, if i recall, did we?
please close it

clement-moulin-frier, 2017-08-14:
Yes indeed, I close it.
Cheers

closed 2017-08-14 by clement-moulin-frier