robotology/mesh-support #514: Left shoulder yaw moves by itself

emilio-cartoni, 2017-07-25:
### Description of the failure
Today we restarted our iCub after a period of inactivity and we found that Joint 2 of the left arm (shoulder_yaw) keeps moving by itself after yarprobotinterface is launched.
In particular, after the calibration is apparently complete, the shoulder yaw makes every few seconds a sudden movement on one direction and then reverts back (i.e. if the arm is at 30 degrees after calibration,  after a few seconds it suddenly goes to 42 degrees, then after one or two seconds goes back to 29, then pauses for a few seconds, then goes to 42 and back again).

### Detailed conditions and logs of the failure
- Using the yarpmotorgui, it is still possible to move the shoulder yaw - if you move it to another position, then it will make these movements around the new position
- In some cases, the calibration phase fails saying that checkGoneToZeroThreshold had timeout on joint 2 while going to zero and then the first 4 joints of the arm are disabled.
- There are appears to be no other ERROR messages in the yarprobotinterface log
(although on one restart we got a a failure to connect to a XSensMtx device)

We attach here all the yarprobotinterface logs we saved (we tried multiple restarts but the problem is still here).

[01 yamrprunlog_25_07_2017_11_01_08.log.txt](https://github.com/robotology/icub-support/files/1172976/01.yamrprunlog_25_07_2017_11_01_08.log.txt)
[02 yamrprunlog_25_07_2017_11_12_51.log.txt](https://github.com/robotology/icub-support/files/1172977/02.yamrprunlog_25_07_2017_11_12_51.log.txt)
[03 yamrprunlog_25_07_2017_11_32_43.log.txt](https://github.com/robotology/icub-support/files/1172979/03.yamrprunlog_25_07_2017_11_32_43.log.txt)
[04 yamrprunlog_25_07_2017_11_56_10.log.txt](https://github.com/robotology/icub-support/files/1172978/04.yamrprunlog_25_07_2017_11_56_10.log.txt)
[05 yamrprunlog_25_07_2017_11_58_41.log.txt](https://github.com/robotology/icub-support/files/1172980/05.yamrprunlog_25_07_2017_11_58_41.log.txt)

julijenv, 2017-08-02:
Hi @emilio-cartoni ,
could you try to get a video of the movement that does the robot, it may help to understand better what happens.
Nevertheless, please try to check the tension of the cable 1 to 8, check that the 2 pulleys beneath the core of the shoulder joint have their screws well tightened.
check the dsensor that you can see on the face of shoulder, try to move a bit the cable and see if the movement happen - keep the red button on your hand when doing so.
let me know,
wainting for a video

emilio-cartoni, 2017-08-02:
Hi @julijenv,
here's the video: 
https://drive.google.com/file/d/0B_hx0oRHclN3WGMydmJFSjVub1E/view?usp=drivesdk

I'll make the suggested checks.

julijenv, 2017-08-02:
Hi @emilio-cartoni ,
is it a video after calibration and without any other module running or do you have another app/module running in background???
Could you check if yarpmotorgui can see the change that happen to this joint and check in the meantime the PID from this joint respect to the right arm same joint.
let me know

emilio-cartoni, 2017-08-02:
@julijenv the video was recorded after calibration, without other apps running (I think only yarprobotinterface and yarplogger were active at that time)

The yarpmotorgui could see the movement (the degrees I wrote in the first post were read from the motorgui).
I'll check the PID.

emilio-cartoni, 2017-08-02:
dear @julijenv , here are the position PIDs from the motorgui

**Left shoulder Yaw:** 
Kp 10000
Kd 0
Ki 10
Position shift factor 13
_PID Output limit 800
PID Integral limit 800_
PID Output offset 0
Positive stiction offset 0
Negative stiction offset 0

**Right shoulder Yaw:** 
Kp 10000
Kd 0
Ki 10
Position shift factor 13
_PID Output limit 1333_ <<<<<-- different from left!
_PID Integral limit 1333_  <<<<<-- different from left!
PID Output offset 0
Positive stiction offset 0
Negative stiction offset 0

Velocity PIDs are all zeros, and Torque PIDs are identical, except for the first variable (Kp) which is -30 on left yaw, +30 on right yaw.

emilio-cartoni, 2017-08-02:
@julijenv Cable 1 and 2 are loose on their pulley (I can move them by hand).
I'll try to tension them again.

julijenv, 2017-08-03:
Hi @emilio-cartoni ,
ok let me nkow

emilio-cartoni, 2017-08-03:
Hi @julijenv ,
I tensioned the two cables again but the problem is still the same.
Sometimes, after the calibration has run successfully (which does not happen often, as it usually fails because of checkGoneToZeroThreshold error on Joint 2 and the first four joint are put in Idle), the arm starts to do its strange movements and then suddenly Joint 2 is put into idle automatically.
If I reactivate Joint 2 from motorgui and make a movement it goes back to idle.
Here's a log recorded while this scenario happened (although at quick glance there seem to be no messages regarding why and when Joint 2 is put to idle).
[log_pc104_yarprobotinterface_1348.txt](https://github.com/robotology/icub-support/files/1196836/log_pc104_yarprobotinterface_1348.txt)

julijenv, 2017-08-04:
Hi @emilio-cartoni ,
could you please tell me if one of the pulley beneath the main shoulder axe is moving or both when the movment occur?
If one please as I ask you before check if the screws holding the pulley are well tightened (take a screwdriver and check).
If this test is not enough, there are 2 possbilities, either a problem in the eletrical connections in the back or mecanical (which rimgs more a bell to me) with the harmonic drive which is loose - to test that could you hold the arm and see if the motor if one only, just seems to spin inside before starting to activate the joint?
You can join me on skype if ever needed.
skype ID : icub.support

emilio-cartoni, 2017-08-07:
Dear @julijenv,
it seems only one pulley is moving, the back one.
See videos:
https://drive.google.com/file/d/0B_hx0oRHclN3cUpaMjFKVnlzQTQ/view
https://drive.google.com/file/d/0B_hx0oRHclN3SU5yZlJxRkpMVkE/view
I tried to tighten the visible screws of both pulleys but they were all already tightened, so nothing changed.
As for the second test, I tried to hold the arm and listen but I didn't hear anything special - not sure if I understood the test however.

I think a more in-depth Skype session will be needed, but everybody (me included) will have a summer break now, so I guess we might arrange it for the last week of August if you are available.

emilio-cartoni, 2017-08-24:
@julijenv 
After the Skype call, I reassembled the pulley and I tried to move it back and forth as you suggested.
At first, I didn't see anything wrong, however after a few tries in different positions I did feel a slight sensation of the pulley sliding a bit on some movements (moving a little bit easier and doing a bit less noise). It was just a slight sensation tough, nothing glaringly obvious (to me).

julijenv, 2017-11-05:
> We disassembled the motor of the pulley and tightened the internal screws as suggested by Julien in a Skype meeting.
> However, the screws seemed to be tightened already, so we probably didn't find yet the real cause of the problem.
>
> We have assembled the motor back and we were then sent a calibration procedure but we are not able to follow it since it gives us an error.
> Calibration shoulder.docx<redir.aspx?REF=D3JPuHGMLIroF5UlPg4YdjfgecOHZiBCOB85xbHxwymb7MIYAiTVCAFodHRwczovL2dpdGh1Yi5jb20vcm9ib3RvbG9neS9pY3ViLXN1cHBvcnQvZmlsZXMvMTQzNzAwNS9DYWxpYnJhdGlvbi5zaG91bGRlci5kb2N4>
> The procedure asks to start yarprobotInterface after setting skipcalib and useencoderRaw to TRUE in the general.xml file.
> After that, we should start the motorgui and read some values there, however starting the motorgui gives us the following error:
>
> ERROR head iencs -> getencoders() failed, retrying ...
>
> If we start the motorgui with only the left_arm part (which is the arm we want to calibrate) we still get the same type of error (with left arm instead of head in the error message).
>
> So we can't follow the calibration procedure.
>
> We need the iCub to be back in operation by December 1st, since we'll have visitors coming who will use the robot.
>
>   1.  When can we have the next Skype session to resolve the issue?
> We have setup our laptop so that it will be possible for you to operate remotely if necessary.
>
>   2.  Is it possible to start the iCub with the left_arm disabled to keep using at least the right arm meanwhile?
>
>   3.  If the issues cannot be resolved remotely, could you send us someone here to repair the iCub?

Hi Emilio,
sorry for not answering you tbefore, I was in Osaka to repair a robot when we ve got a bit of a talk over skype, but with the time difference. I had to leave you. Afterwards I was for a weel in holidays in Japan, right now Im in Korea waiting for getting my flight to retun in Italy.
I'll be all day long in front of my desktop tomorrow (meaning on monday).
Please contact me,
earlier the better.
Or give me a date to get in touch on skype.
let me knwo

 [Description: iit-logo-icub]

Ing. Jenvrin Julien aka iCub Doctor
Fondazione Istituto Italiano di Tecnologia
iCub Facility
Via Morego, 30
16163 Genova
Tel.  : 01071781453
e-mail: julien.jenvrin@iit.it<https://mail.iit.it/owa/redir.aspx?C=5O2KFbeABEyD_6VNPoGy3F87799dW9III0QVopPe7CFNwYcn3hDAc_xDBzgAWdjnuNDrN4pwpLs.&URL=mailto%3adavide.tome%40iit.it>
www.iit.it<https://mail.iit.it/owa/redir.aspx?C=5O2KFbeABEyD_6VNPoGy3F87799dW9III0QVopPe7CFNwYcn3hDAc_xDBzgAWdjnuNDrN4pwpLs.&URL=http%3a%2f%2fwww.iit.it%2f>

P Please consider the environment before printing this email

julijenv, 2017-11-05:
Hi @emilio-cartoni ,
sorry for not answering you tbefore, I was in Osaka to repair a robot when we ve got a bit of a talk over skype, but with the time difference. I had to leave you. Afterwards I was for a weel in holidays in Japan, right now Im in Korea waiting for getting my flight to retun in Italy.
I'll be all day long in front of my desktop tomorrow (meaning on monday).
Please contact me,
earlier the better.
Or give me a date to get in touch on skype.
let me know

closed 2018-01-16 by julijenv