robotology/mesh-support #515: Left arm quadrature encoder broken

S-Dafarra, 2017-07-26:
When running the robot we got the following error
``
26383	1112,195909	ERROR	 from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 1423s 635m 576u: (code 0x0200000d, par16 0x0003 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
``
Through the motorgui we were able to run the motor, but as soon we try to move it, it goes in HW FAULT.

@julijenv @DanielePucci @gabrielenava @pi-q

traversaro, 2017-08-10:
It seems that the problem is still present. 
@S-Dafarra @nunoguedelha @fjandrad

DanielePucci, 2017-08-10:
@julijenv @spada1978 can any of you help us with this blocking issue?

Thank you guys.

julijenv, 2017-08-11:
Hi @S-Dafarra , @traversaro , @DanielePucci , @nunoguedelha ,
I opened the whole back cover and check everything, the sensors work perfectly( HALL from motor and LCORE - optical device), I disconnected the motor and run it with my debug tool, and I'm sure it works fine.
I rewired and changed the LCORE connector . I dont know what else to do or from where it is coming from.
Perhaps the 2FOC board has some issue - I will have to make a test by changing it.

DanielePucci, 2017-08-11:
Is it possible that, somehow, the 2FOC problems cause this issue (i.e. `Left arm quadrature encoder broken`)? 

Emiliano was mentioning that electrical problems may be occurring in the arm, which in turn may cause the issue. What do you think? @julijenv

nunoguedelha, 2017-08-17:
@julijenv I'm so sorry, but after all your efforts to fix it, it worked for some time, but just started failing again, systematically.

julijenv, 2017-08-21:
Hi @nunoguedelha , @DanielePucci , @S-Dafarra , @traversaro ,
From my last debugging, it looks like that this issue finally went away after having changed the sensor LCORE.
It has ben taken from warehouse
IITCODE Qta Description
6506 x 1: LCORE

closed 2017-08-21 by julijenv