robotology/mesh-support #517: icub left_arm's finger (j13-14) get stuck when I open the arm

misaki43, 2017-07-26:
Hello,

In my experiment, I close / open the left arm of the robot. At one trial, when the robot had to open its hand, the middle finger goes in the other direction and got stuck in a totally "close position".

On yarpmotorgui we can see that even thought the order position of the joint 13 was something like 7°, it get stuck in bigger value (e.g. in yarpmotorgui, the position was on the right of the bar even thought the order was on the left).

Dorian fixed it by turning a screw on the hand, and when I re-launched the robot it worked fine again during some trials. Then, the problem appears again.

Here is the log session, what can I do ?
[log_pc104_yarprobotinterface_965.txt](https://github.com/robotology/icub-support/files/1176631/log_pc104_yarprobotinterface_965.txt)

Thank you for your help!

@serena-ivaldi @dogoepp @olivierrochel-inria 

Oriane

dogoepp, 2017-07-27:
What we don't understand is why the motor actuating the finger would move in the wrong direction (increasing the error).

The finger's mechanics seems al right, since it can be folded and unfolded manually (when motors are of, by using a screwdriver on the motor's "shaft"). Should I be looking for the an encoder or for something more software-related ?

Besides, we observed that the first joint of the middle finger of both hands is a bit inwards :
[image]

emilio-cartoni, 2017-07-31:
@dogoepp maybe the tendon got too much loose (totally unwinded) and now it's winding around its pulley in the wrong direction?
When you turn the motor with the screwdriver and turn clockwise, what does it do?
Is it the same as the motor for another finger? (i.e. they both close or they both open when turned clockwise or does the motor of j13-14 behave differently)

julijenv, 2017-08-02:
Hi @misaki43  and @dogoepp ,
Could you tell me what is the value of the stretch finger for both joint 13-14, thx in advance.
if not zero, please try to move the magnet in order to get 0 (or close) when stretch.
let me know

julijenv, 2018-01-16:
Still waiting for feedback on this

closed 2018-01-16 by julijenv

dogoepp, 2018-02-28:
Sorry again for the radio silence. I get to work with iCub only when others use it.

It seems that the problem was "temporary". It suddenly disappeared.