robotology/mesh-support #518: Left arm goes on hardware fault at any movement

tanismar, 2017-08-01:
Please **fill in the following two sections** and then **remove any comment and/or remark become unnecessary**:

### Description of the failure

Every time I try to get the robot to go to the home position, the left arm goes idle and on yarpmotorgui shows hardware fault on joint 6, and idle on joint 5. 

### Detailed conditions and logs of the failure
Log shows on robotInterface:
78645	3062,219161	ERROR	sharedcan [6] board 8 FAULT OVERLOAD AXIS 1
78643	3062,209080	ERROR	sharedcan [6] board 8 FAULT OVERLOAD AXIS 0
78660	3062,481293	ERROR	Unable to set the control mode of a joint ( j:6) in HW_FAULT

On gravity compensator: 
73341	2872,778017	ERROR	gravityCompensatorThread lost connection with wholeBodyDynamics.
73342	2872,778029	ERROR	gravityCompensator module lost connection with iCubInterface, now closing...

---
**Remark**: if you are not a daily icub-support user, please read carefully the guidelines for contributing linked at the top of this page. In particular, the **name of your robot** must be always indicated: either using **labels** if you're already a contributor, as per standard, or by specifying it while detailing the issue.

julijenv, 2017-08-14:
double ticket same as #516.
Could you please recheck, I do believe I clear this one, thx in advance.

tanismar, 2017-08-14:
Great, I'm home today thought. If there is anyone around, please ask them to chekc (Massimo, Phuong, Giulia... ). Otherwise I'll check on Wednesday.

julijenv, 2017-08-14:
no one around, whenever you will return ;)

julijenv, 2017-08-21:
@tanismar ,
did yu check ???

tanismar, 2017-08-21:
Hi Julien.
Nope, the system was up on Thursday morning, but everything needed to be
reinstalled before being able to test the robot, and I wasn't able to
complete the installation. On Friday I left on holidays and I'll be back on
Monday 28th,so until then I can't test anything. Ask maybe Giulia or Phuong
to test it if they are around.

Best,
Tanis

julijenv, 2017-08-22:
hi @tanismar ,
what specifically do u launch to be able to get this error message???
let me know the procedure, in order to us to test.

julijenv, 2017-08-22:
Hi @tanismar ,
from what I heard from Phong, it is that there was an issue with whoolebodydynamics parameters BUT with Giulia they make it worked.

tanismar, 2017-08-22:
Hi, 
What I did when we tested it together was the following:
Turn on the robot, launch robot interface and all the controllers of icub_startup, then run actionsrederingengine. 
Then I order to generate some movements, I opened a rpc port
$yarp rpc /actionsRederingEngine/cmd:io
And send the home command
>>home all 
Or the expect command to make it move in another way:
>> expect
We also sent it home using the robotmotorgui home command, and then looped between those 3
2-3 movements. 

Hope it helps. 
Best, Tanis

closed 2017-10-13 by julijenv