robotology/mesh-support #521: General questions about fixing the robot

mulkkyul, 2017-08-07:
Hi,
This is jungsik from KAIST. (iCubDaejeon01)
I have general questions about fixing the robot.

1) Is it possible to send our icub to IIT and get it fixed?
2) Is it possible to invite a technician from IIT to our place? (our place will be Okinawa, Japan).

### Description of the failure
- Unstable booting procedure (#160, #278) 
- Freezing while opening CAN devices (#279) 
- Short-circuited tactile board (#275) ; We got the new tactile board but it hasn’t been installed yet.
- Misplaced Tendons (fingers)

Thank you.
jungsik

julijenv, 2017-08-14:
Hi @mulkkyul ,
the 2 first issues are quite big one to be solved for sure to be able to continue using your robot, please refer to my reply to #279 .
for the 2 last ones I think we can solve this easily with skype.
By the way before proceeding by sending the robot to us or someone coming I think a skype meeting will be much appreciated.
my skype account Id is: icub.support
let me know ASAP you got to try my last reply

julijenv, 2017-10-10:
Hi @mulkkyul ,
did you perform the ladt test I was ssuggested you on #279 ?
did you try to get in touch with andrea.derito@iit.it or giorgio.metta@iit.it for sending your robot to GEnova?
let us know, I didnt hear from you since my last reply.

closed 2019-10-12 by pattacini