robotology/mesh-support #522: Left elbow broken

anatanevska, 2017-08-08:
### Description of the failure
Left elbow broken

### Detailed conditions and logs of the failure
Tendon 9 of red iCub (iCubGenova03) snapped and needs to be replaced.

amaroyo, 2017-08-09:
@julijenv, adding few more things:

left thumb is loose and the cable gets stuck some times not allowing the thumb to move. 

Left flat cable of the eye seems about to break. When you move the eyes to the left, you begin losing vision seeing blue horizontal lines, till you lose totally the vision.

julijenv, 2017-08-14:
Hi @amaroyo please create a new issue ticket for your last comment, it isnt part of left elbow broken, and will be misleading to icub-support team.

closed 2017-08-16 by julijenv