robotology/mesh-support #526: iCubBielefeld03: firmware status

lschilli, 2017-08-16:
I am currently updating the software of iCubBielefeld03. I have a few questions regarding the currently installed firmware:
* Is the currently installed firmware recent enough for yarprobotinterface?
* Can I just update the firmware using `updateRobot.sh` or are there problems to be expected due to the age / version of our robot?
 
The following screenshots show the currently installed firmware versions:  
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

lschilli, 2017-08-18:
Update: Currently installed firmware works with current yarprobotinterface for torso and head. In any case a quick check of the current firmware versions will be appreciated.

julijenv, 2018-04-09:
Hi @lschilli ,
sorry for answering so late, but well I knew that as long as you dont try to launch the robot you will never know.
from the 9 printscreens you sent me just the one of the line 3 left leg is strange.
[image]
except that I guess that the upper part of the robot is ok, the only part you use I am sure.
let me know and I will let you close the 2 issues you opened for the same pseudo problem you absolutely knew how it works ;)
ciao Lars,

lschilli, 2018-04-09:
Well, the idea was to ask for a quick check via this issue and then close it. The latency was a bit surprising but finally I can assume they are not that much outdated ;) Thanks for pointing out the leg issue.

closed 2018-04-09 by lschilli