robotology/mesh-support #529: left and right arms inertial MTB boards init configuration failure at startup

nunoguedelha, 2017-08-28:
### Description of the failure
After activating the MTB boards of the arms (in the xml config file loaded by **yarprobotinterface**), and when starting the **yarprobotinterface**, the configuration of the MTB boards in `embObjInertials` fails.

### Detailed conditions and logs of the failure
File loaded by yarprobotinterface: `icub_all_inertials.xml`
Added lines for activating the MTB boatds:
```
    <!-- INERTIAL SENSORS ON MTB -->
    <devices file="wrappers/inertials/left_arm-inertials_wrapper.xml" />
    <devices file="hardware/inertials/left_arm-eb24-j4_7-inertials.xml" />
    <devices file="wrappers/inertials/right_arm-inertials_wrapper.xml" />
    <devices file="hardware/inertials/right_arm-eb27-j4_7-inertials.xml" />
```

We get the error message:
```
958621	420505,770841	ERROR	embObjInertials::fromConfig() is using a non supported IP address: 10.0.1.24
958623	420505,770855	ERROR	embObjInertials missing some configuration parameter. Check logs and your config file.
958636	420505,780905	ERROR	embObjInertials::fromConfig() is using a non supported IP address: 10.0.1.27
958638	420505,780913	ERROR	embObjInertials missing some configuration parameter. Check logs and your config file.

```

So apparently, IP addresses 10.0.1.24 and 10.0.1.27 are not supported.
@marcoaccame it seems that we already had this issue in the past but if I remember well it had been fixed, right?

marcoaccame, 2017-08-28:
Hi @nunoguedelha,
we found this problem for board EB22 on which on new robots there are attached the MTBs of the chest. And I added the case of IP address 10.0.1.22: https://github.com/robotology/icub-main/commit/575e722fc17f0bc34bdc24d1d0042cc2fc68f137

However, on the robots with the new arm controlled by MC4PLUS boards, the MTBs on arms are on new IP addresses 10.0.1.24 and 10.0.1.27 which are not supported yet.

It is easy to add them inside embObjInertials.cpp: I will tell you when I have done it.

marcoaccame, 2017-08-28:
Hi @nunoguedelha, I have added support for boards .24 and .27 with https://github.com/robotology/icub-main/commit/202fe8bb0e48140ccc46294d458df0d98cae5506.

It will work now.

nunoguedelha, 2017-08-28:
Thanks @marcoaccame ! I will test tomorrow and close this issue right after.

marcoaccame, 2017-10-17:
Hi @nunoguedelha, I suppose it is time to close the issue.
But reopen it if it is still not ok.

closed 2017-10-17 by marcoaccame

nunoguedelha, 2017-10-18:
Yes sure, I'll check again as soon as I have access to the robot. Thank you.