robotology/mesh-support #530: Gaze malfunction (oscillation) while fixating a point

vicentepedro, 2017-08-30:
Hi all,

We are experiencing a strange behavior in our robot when using the Gaze Interface.

We send one fixation point (a 3D Cartesian point which is possible to gaze at by the robot), and the controller starts a _quasi_-infinite loop until finally converging to a solution. You can see an example of the behavior in this video:
https://youtu.be/vi1zRrbi5So

The above behavior seems to be head configuration dependent. It seems to be related to the coupled joints of the neck (pitch and roll).

We checked the hardware. The steel cables are stretched and the electronics (without a rigorous inspection) seems also fine. The only thing that we notice is a backlash on one of the coupled joints (the right one) as you can see here:
https://youtu.be/1pywYkeE3Pw

The backlash seems to be on the harmonic drive. And, if we understand correctly, the motor encoder (before the harmonic drive) and the absolute encoder (after the harmonic drive) could be reading incongruent joint angles. Am I right?

Could this backlash be sufficient to trigger this gaze behavior? Or is it more a software problem? We have not changed the head PID values for a few years.

/cc @gsaponaro @NunoDuarte

julijenv, 2017-08-30:
Hi @vicentepedrodro ,
to me it isnt clear 100% that the problem is coming from the harmonic drive, for 2 reasons:
1 - this shaking behaviour appears just when you launch the ikinGazeControl
2 - the shake is quite just of [-1;1] from what I can see on the video.
If I were you, I would try to move the head just with the yarpmotorgui (without iKinGazeControl) and go from a position to another (try to get the postiotion you showed me in the video) and let me know if whether or not the head is dangling.
If not, I would guess that there is no problem with the coupled joint 0 and 1 at all. BUT if you feel more releave by making a clean check of the overall screws and tighten them it is never a bad idea.
You may possibly think that the encoder as eretic values, try to insulate the encoder with some cooking aluminium sheet of paper and see if it prevents this behaviour.
The movement looks to be the joint 1 related, is it possible to check the screws at the level of the neck
[image]
[image]

Maybe @pattacini could give his point of view of the little dangling and point out one specific parameter inside the ini file of the gaze controller that could affect this.
Let us know,

pattacini, 2017-08-30:
Hi @vicentepedro 
You said that it's configuration dependent. Thus, does the dangling occur also with the head in the upright position (i.e. looking at e.g. `-0.5 0.0 0.36` in Cartesian space)?

vicentepedro, 2017-09-01:
Hi @pattacini, 
After a few more tests, we believe that the problem is not configuration-dependent (we tried different 3D positions), but that it depends on having the **tracking mode** of the gaze on or off.

When we send a look command in the upright position (e.g. -0.5 0.0 0.36), the dangling behaviour appears consistently when the tracking mode is on. This happens even when the robot is still in the initial position (initial position after yarprobotinterface calibration).

pattacini, 2017-09-01:
Thanks @vicentepedro for the further tests that helped narrow down the problem.

What about the **stabilization mode** instead? If you don't mind I'd ask you to repeat the same bunch of experiments you already did with the stabilization mode disabled and with the tracking enabled.

This only if the stabilization was on in your previous tests, of course.

vicentepedro, 2017-09-01:
The stabilization mode was off in our tests.

pattacini, 2017-09-01:
Ok, thanks @vicentepedro 
I'll peruse the code devoted to handle tracking mode and get back to you once done.

pattacini, 2017-09-04:
Hi @vicentepedro 

I suspect that the code lines involved are, in order:

https://github.com/robotology/icub-main/blob/01d8c4a0bc24c5ce56d31232fef92cce71da56eb/src/modules/iKinGazeCtrl/src/solver.cpp#L771

https://github.com/robotology/icub-main/blob/01d8c4a0bc24c5ce56d31232fef92cce71da56eb/src/modules/iKinGazeCtrl/src/solver.cpp#L791-L792

where:
https://github.com/robotology/icub-main/blob/01d8c4a0bc24c5ce56d31232fef92cce71da56eb/src/modules/iKinGazeCtrl/include/iCub/solver.h#L40

It might be that a threshold of `0.1 deg` for detecting changes of the torso joints is too low for some robot configurations.

I gave it a try on last Friday on our Blacky and we have very small oscillations (due to the recomputation of IK) compared with your case, up to the point where they get extinguished.

Nonetheless, there's really no advantage in keeping such a low threshold on the torso joints to retrigger gazing. Therefore, I've just committed a new threshold onto `devel`, which is now `1.0 deg`.

Could you test it, please? As of now, there are no available time slots for me to go on our robots.

pattacini, 2017-09-04:
Also, it could be helpful logging the output of `/iKinGazeCtrl/events:o`.

vicentepedro, 2017-09-04:
Thanks @pattacini 
As soon as I test it on the robot I will let you know.

vicentepedro, 2017-09-05:
Hi @pattacini 
We already changed the threshold to `1.0 deg`

The dangling is smaller now. Almost unnoticeable or absent. 
However, we can see that the motors are changing their position even when the robot is still and the fixation point is unchanged.
https://youtu.be/6nUfzHjjVw0

I don't know if this behavior is expectable or not.

Another thing that we have observed today is when the tracking mode is on and we move the torso pitch, the neck will not update its position, but the eyes will (i.e., the robot tracks the point only with the eyes). In this situation, when the torso is very bent forward, the 3D point becomes "invisible" and not in the center of the image anymore. This would not happen if the neck were used.
(I also don't know if this is the expectable behavior of the iKinGazeCtrl) 

Thanks again for your patience @pattacini ;)

pattacini, 2017-09-05:
@vicentepedro bear in mind that there exists a gear ratio between the motors and the joints, therefore the motors keep on rotating under the mere action of PID controllers. In a nutshell, the motors rotate while the joints don't, as there's a gear reduction.

Thanks for spotting out the bug on the torso pitch 👍
Your observations led me to better reconsider the code - which is by the way quite complex 😉 

We need a robust estimation of the torso velocity from position samples to catch the fact the torso has moved. Hence, I've pushed a patch on `devel`.

Probably, we have now to tune the threshold acting on the velocity...

pattacini, 2017-09-11:
@vicentepedro did you have the opportunity to test the latest patch?

gsaponaro, 2017-09-13:
Hi @pattacini
We have not tested it yet. Due to some demos and the upcoming ICDL-EpiRob conference here, we can only look at this in the last week of September. @vicentepedro and I are closing the issue for now, and we'll let you know later any comments, OK?
Thanks a lot :)

closed 2017-09-13 by gsaponaro