robotology/mesh-support #532: Hardware problem in the left knee

dogoepp, 2017-09-06:
### Description of the failure
Hardware problem in the magnetic position sensor of the left knee.

### Detailed conditions and logs of the failure
When trying with @InriaBrice to do a squatting motion with the iCub we get the following error:
 
> 82131	1733,719156	ERROR	 from BOARD 10.0.1.6 (left_leg-eb6-j0_3), src LOCAL, adr 0, time 1934s 596m 521u: (code 0x02000008, par16 0x0003 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .

After this error, the following comes up repeatedly:

> 82134	1733,729253	ERROR	setReferenceRaw: skipping command because BOARD left_leg-eb6-j0_3 IP 10.0.1.6  joint  3  is not in VOCAB_CM_POSITION_DIRECT mode

It appeared twice in a row. Inbetween we had restarted `yarprobotinterface`. However, this error does not come up if iCub is on its pole.

For these experiments the joints were in position direct mode. Also the interaction mode is compliant. Values used for interaction mode:

- stiffness: 0.111
- damping: 0.014

dogoepp, 2017-09-06:
After stopping also the motors (switch them off), the following error comes right at startup of `yarprobotinterface`:

> ERROR	 from BOARD 10.0.1.6 (left_leg-eb6-j0_3), src LOCAL, adr 0, time 79s 419m 472u: (code 0x05000016, par16 0x4008 par64 0xffffffffffff3000) -> CFG: EOtheEncodeerReader cannot be configured: verification fails. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .

marcoaccame, 2017-09-06:
Hi, the diagnostics message tells that the fourth joint (j3) has a problem with its primary encoder (which is an AEA). The reported error code is *encreader_err_AEA_CHIP* (which is = 3). All other encoders report *encreader_err_NONE* (which is = 0), hence are working fine.

An error code *encreader_err_AEA_CHIP* means that the AEA chip communicates correctly over SPI, its data is validated with parity check, but the chip tells that it has had a problem in the acquisition of data and hence the reading is not valid.

I believe that in such cases @julijenv changes the AEA chip.

julijenv, 2017-09-06:
Hi @dogoepp ,
actually I think that if you just takea  look at the left knee of the robot you should see a broken wire for either inside or outside of the knee mechanical (sensor).
take a look and repair it should rework, fairly after that.
let me know

InriaBrice, 2017-09-07:
Hi, is it possible to get a list of possible error and their meaning ? Like a little manual of reparation when an error occurs. It would be great for us.

Best regards

dogoepp, 2017-09-07:
Hi @julijenv,
It seems that you were quite right. The cable 5S1, attached to EB6 is badly broken, as shown on the picture bellow.

[image]

Do you have an idea how it broke like that ? More importantly, Where I should source the replacement wire ? It seems that I have to redo the whole cable.

Also, there seem to be encoders on each side of the knee. Could you remind me which one is for what ?

julijenv, 2017-09-07:
Hi @dogoepp ,
From what I can see if you have so much slack in the cable, to shorten it should not be such a problem, just cut it the level of the pinch and resolder it directly at the level of the board (this one is the inside one of the knee- odds are that if it is pinched it is the one which trouble your log messages).
FIrst thing first so, try to repair this visual problem and resstart the robot and see if it is this one or not - but I do believe that you found the guilty encoder.
To answer at your question about what does each encoders on the icub (AEA mainly for a specific joint like the knee for example) we will need some help from the software developpers to have more inside and precision about it but from what I know and the positionning of them, I would guess that the inside one - is given the angle of the knee, main angle and by consequence main AEA encoder for this joint - the other one (outside one) should be coupled with another encoder to give the deformation of the ring (if you use it, if you dont, basically it should give you the same value as the inside encoder). The speed should be given by the LCORE sensor, sensor right above the one you pictured which read the internal disk).
But let's make answer to that the oppotunate person. @marcoaccame .
To @InriaBrice , I think some kind of a little manual should be released somehow to explain more in details all the different error messages that could be printed in the log file and how to handle and check and verify the specific encoder or hardware failure. The self explanatory line comment added can drive you to the solution but it is quite short to understand by yourself what happend and that 's why you need us to read it and give you hint, but soon you will be able to debug by yourself with the help of this manual.
More details soon,

dogoepp, 2017-09-07:
Thanks a lot Julien! Following your guidance we could fix the issue. However, we could not solder as well as it was originally done. I fear that we could have later issues. Also, we had no more lack left and the red cable ended up too short. It will probably run loose sooner or later.

Is there some trick for better soldering results ? We might need to replace a full cable someday. What is the best approach. Should we order the same cable (from where ?) and two-three sensor boards (in case one is broken from too much re-soldering) ?

Seing the picture bellow (glue was later added), I think we might as well consider to replace the whole sensor as preventive maintenance. What do you think of it ?

[image]

Also, what heatshrink to you guys use ? The only one I had that would fit the cable does not shrink as much as the original one.

julijenv, 2017-09-19:
Hi @dogoepp ,
Unfortunately, the board is our own product and the cable is specifically sold by industrifil and I dont know if it is easy to get some of their product at lest to make a big order.
for the heatshrinking tube, i can give you some reference you can fetch easily on farnell or RS.

closed 2017-10-10 by julijenv