robotology/mesh-support #533: Timestamp missmatch from left_arm/stateExt:o and the rest of the  stateExt:o ports

fjandrad, 2017-09-06:
### Description of the failure
When reading the timestamp provided by the ports 
/icub/left_arm/stateExt:o 
using :
 yarp read ... /icub/left_arm/stateExt:o envelope 

there is a time mismatch of around 300 seconds with respect to the other stateExt:o ports
[image]

Could you help us understand why? and possible fix it ?
@marcoaccame @barbalberto

traversaro, 2019-04-10:
Is this problem still there? Somebody that is currently using the robot can quickly check?
cc  @kouroshD @MiladShafiee @Yeshasvitvs @lrapetti

fjandrad, 2019-04-11:
I think it used to happen randomly, I have not seen it much lately though.

closed 2019-10-12 by pattacini