robotology/mesh-support #536: Backlash on ankles

francesco-romano, 2017-09-11:
The robot has big backlash on both ankles.
This may create oscillations when the robot is standing.

@spada1978

francesco-romano, 2017-09-12:
It seems that the backlash on the right ankle is due to some stripped screws.
On the left instead, it seems that the problem is in the motor / transmission.

spada1978, 2017-09-12:
Per quanto riguarda la caviglia sx si necessita il controllo del giunto RC_IIT_018_G_003
Mentre per la caviglia dx si necessita di filettare il foro del pezzo RC_IIT_018_P_047 in quanto la vite gira a vuoto.
Appena francesco-romano ultimerà gli esperimenti prenderemo in consegna il robot per le riparazioni di entrambe

francesco-romano, 2017-09-12:
cc-ing @ChiaraBartolozzi @vvasco @arrenglover @Iaxama as they will be affected by the maintenance.

julijenv, 2017-09-19:
Hi @francesco-romano ,
this morning @spada1978 repaired both joint 5 for Left and Right knee, seems ok to me, now @valegagge is recalibrating the entire robot, if she cant find any issue related to the knee I'll close this issue.

valegagge, 2017-09-25:
I recalibrated all joints with absolute encoder connected to ethernet boards. All is ok. Both joints move correctly, but i didn't performs other tests.

Before closing this issue, it should be better to verify that robot hasn't backlash on ankles.

closed 2017-10-10 by julijenv