robotology/mesh-support #540: Broken cable on iCub Head and Broken pinky (right_arm)

gonzalezJohnas, 2017-09-22:
### Description of the failure
iCub head broken cable.
[image]

iCub pinky ( right_arm ) close but don't open.

### Detailed conditions and logs of the failure
During experiments we had uncontrolled behaviors resulting on breaking one cable for the iCub head.
Concerning the iCub pinky, the encoder works displaying the right value, but  only to close the pinky, the open movement no longer work. 

---

julijenv, 2017-09-22:
Hi @gonzalezJohnas ,
could you please , check if the pulley next (iniside neck) from the serial number SN034 has  one single piece with 2 screws on it or if there are 2 set screw. to see it just spin the external pulley until you make the inside pulley make a turn,
let me know

gonzalezJohnas, 2017-09-22:
This is the pulley you are talking ? 
[image]
[image]

Then there is only 2 screws on it.

julijenv, 2017-09-26:
IT has been used 
IITCODE Qta Description
5965 x 1 : cable A neck

closed 2017-09-26 by julijenv