robotology/mesh-support #541: Left shoulder tendon replacement + calibration

gsaponaro, 2017-09-22:
During a manipulation experiment, we broke the left shoulder tendon. Our technician is in the process of replacing it, but we have a couple of questions/remarks below.

1)
The metal piece that supports the arm (in green in the pictures below) seems rotated counterclockwise with respect to the pen markers (highlighted in red):

[image]

[image]

We suppose that we have to rotate the metal piece back so that the markers will be aligned as in this picture (in blue in the picture below), correct?

[image]

2)
After we have remounted tendon and arm, what is the recommended calibration procedure on the software side? In the Manual I see these documents:
- [Hand Arm Calibration Procedure for iCubV1.1-2-3](http://robotcub.svn.sourceforge.net/svnroot/robotcub/trunk/iCubPlatform/doc/manuals/ST2009_RBCS_ICUB_ARM_HAND_CALIBRATION_01.doc)
- [Arm fine calibration](http://wiki.icub.org/wiki/ArmFineCalibration)

Anything else that we should be careful about? Of course we will do the calibration with the fault button pressed. Thanks in advance.

julijenv, 2017-09-25:
Hi @gsaponaro ,
you have the new shoulder design and this one allows you to do not break it if ever your arm at some point in your experimentations touch the robot or a hard surface, the joint slide as a protection.
Consider doing the fine calibration, because you have just few degrees changed and basically just on j[0].
let me know,

closed 2017-09-28 by gsaponaro

gsaponaro, 2017-09-28:
Thanks @julijenv , we have re-calibrated j0 which was off by 25 degrees, and we also tweaked j2 j3 in the process.