robotology/mesh-support #542: Icub Head calibration issues iCubGenova03(red)

gonzalezJohnas, 2017-09-26:
### Description of the failure
The iCub head shaft is curved, there is a play in the right eyes due to the central piece linking the two eyes,
and right camera is not aligned with the white bulbe.
This issue cause a bad calibration of the iCub eyes that need to be redo.

### Detailed conditions and logs of the failure
After fixing of the broken cable of the iCub head we saw this issues.

julijenv, 2017-09-26:
It has been requested
IITCODE Qta Description

2634 | RC_IIT_010_P_055 |   |   | Custom mechanical parts (metal, polymers, etc.) | V2 eye tilt shaft

julijenv, 2017-10-13:
it has been replaced the part inter-eye movement (IIT OCED 2634), the face has been completeed unmount to be remount later, the T shape was bent and I had to reput it staight.

closed 2017-10-13 by julijenv