robotology/mesh-support #543: Left arm board failure on iCubErzelli01

ddetommaso, 2017-09-27:
### Description of the failure

During the execution of the yarprobotinterface a failure on a left arm's board occurs. The robot do not move at all at the start-up.

### Detailed conditions and logs of the failure

Please below you can find the related errors of the yarplogger 

ERROR	EthResource::verifyBoardPresence() DID NOT have replies from BOARD right_arm-eb28-j8_11 with IP 10.0.1.28  even after  20  attempts and 10.003 seconds: CANNOT PROCEED ANY FURTHER

ERROR	EthResource::verifyEPprotocol() cannot verify BOARD right_arm-eb28-j8_11 with IP 10.0.1.28 : cannot proceed any further

ERROR	embObjMotionControl: failed verifyEPprotocol. Cannot continue!

ERROR	EthResource::serviceCommand() failed an acked activation request to BOARD right_arm-eb28-j8_11 with IP 10.0.1.28 after 3 attempts each with waiting timeout of 0.5 seconds

julijenv, 2017-09-28:
Hi @ddetommaso ,
Sorry to ask it again but could you please send me the ENTIRE logfile for the session becasuse it is weird that the left arm is not responding if actually in the error message it is the right that show some misbehaviour. ASAP you give me the log I will answer what to try to do by EDL because unfortunately I wont be available for 1 week .
let me know

ddetommaso, 2017-09-29:
Hi @julijenv ,

sorry you're right, I referred the wrong arm. he failure is in the right one. I fixed the issue title and description. Please attached you can find the logfile.

Thank you again.

[log_pc104_yarprobotinterface_1640.txt](https://github.com/robotology/icub-support/files/1343483/log_pc104_yarprobotinterface_1640.txt)

julijenv, 2017-10-09:
Sorry for this very late reply, I was on duty abroad,
This week I will pass by Erzelli, but if you want someone to check something on it, it isnt a big problem, from the erro message it seems that the board 27 is not responding, could you try to open a command shell and check with ethLoader if you can see it.
if not you will have to check the wiring in between board 26 and 28.
[iCub2.5-E3.2.1_5672_Logic.pdf](https://github.com/robotology/icub-support/files/1368027/iCub2.5-E3.2.1_5672_Logic.pdf)
Please have a look at this file. to understand how and where are the boards inisde your robot.

julijenv, 2017-10-13:
For the repair , it ahas been needed
IITCODE Qta Description
5443.D x 1 : MC4-PLUS, IIT - Electronic board with cortex M4, Ethernet and power driver for 4 DC motors, SPI interface, 2 analog channels, 12-32V supply voltage, low brightness
I replaced the brokjen S/N:087 - EB28 by S/N:133.

closed 2017-10-13 by julijenv