robotology/mesh-support #545: embObjMotionControl::open() fails to start mc service for board EB3

dcam0050, 2017-10-06:
### Description of the failure
Running yarprobotinterface to start up the icub I encounter the error `embObjMotionControl::open() fails to start mc service for board EB3` which leads to a host of other failures. 

### Detailed conditions and logs of the failure
I'm not sure where to start debugging the issue. The ethboard EB3 at 10.0.1.3 is alive and discovered normally by ethLoader and the can board status in canLoader is Running for all boards. I've so far tried restarting the icub multiple times and also a complete power cycle but to no effect

dcam0050, 2017-10-06:
This is the complete log for yarprobotinterface 

[yarprobotinterface_log.txt](https://github.com/robotology/icub-support/files/1362525/yarprobotinterface_log.txt)

marcoaccame, 2017-10-06:
The EB3 boards cannot find its 2FOC boards on CAN1:
```
[ERROR] from BOARD 10.0.1.3 (EB3) @ 42s 734m 867u: CAN discovery after 2354 ms 
has detected 4 missing eobrd_foc boards in CAN1:
[ERROR]1 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:1
[ERROR]2 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:2
[ERROR]3 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:3
[ERROR]4 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:4
[ERROR] from BOARD 10.0.1.3 (EB3), src LOCAL, adr 0, time 42s 734m 989u: (code 0x0500000e, 
par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMotionController cannot be configured. 
can discovery of 2foc boards fails. see CANdiscovery messages for more details + .
```

Please ask advice to @julijenv for how to proceed.

dcam0050, 2017-10-06:
@marcoaccame Thanks for the insight. 

@julijenv  How do I proceed?

dcam0050, 2017-10-06:
I updated yarp to 2.3.70.1 and icub-main and robots-configuration to 1.8.0 and I tried running `yarprobotinterface` again but with no success because I need to update firmware. So I reverted the previous installations I had backed up and  when I ran `yarprobotinterface` to check the previous error was still there this time the iCub boots OK

dcam0050, 2017-10-09:
The issue is however back again. @julijenv Any idea as to how I can debug this?

julijenv, 2018-07-19:
hi @dcam0050 ,
Sorry to disturb after so long, IS this still occuring?!?!

dcam0050, 2018-08-23:
Duplicate of #660

closed 2018-08-23 by dcam0050