robotology/mesh-support #554: iCubGenpva04 went in hardware fault while torque controlled

gabrielenava, 2017-10-19:
Together with @fjandrad I was performing experiments on iCubGenova04 by running YOGA++ demo. While the robot was back on two feet (end of the demo), but still torque controlled, suddenly all torque controlled joints went in hardware fault. Furthermore, the yarplogger crashes while we were looking at it and trying to debug the issue. I opened the yarplogger again and this is the output:

[log_icub-head_yarprobotinterface_13142.txt](https://github.com/robotology/icub-support/files/1398455/log_icub-head_yarprobotinterface_13142.txt)

I don't know if this logger may be useful or not for debugging the issue.

ale-git, 2017-10-19:
It is likely a network congestion/delay problem. The torque feedback from the wholeBodyDynamics module can't reach the control boards within a 100 ms timeout, so the closed loop torque control can't be performed because of lack of feedback.

julijenv, 2017-10-20:
close because it is the copy of #554

closed 2017-10-20 by julijenv

randaz81, 2017-10-20:
As mentioned by @ale-git, this is the desired behavior.
Please note that if the SW does not put intentionally the joints in FAULT, the joint tendons/limits would be seriously damaged. The fault is not automatically cleared because we want the user to investigate the source of this network dangerous delay before continuing the experiment.