robotology/mesh-support #555: Segmentation fault at startup

gabrielenava, 2017-10-19:
Debugging [this issue](https://github.com/robotology/icub-support/issues/554), I completely restarted the robot after reverting `icub-main` to `devel` (it was in another branch because of [this PR](https://github.com/robotology/yarp/pull/1396)). I restarted both motors and `icub-head`. I didn't open the logger, and I run `yarprobotinterface --config icub_wbd.xml` directly on a terminal on `icub-head`. The robot didn't start because of segFault problem. See the attached output on `yarprobotinterface` terminal.

[yarprobotinterface_terminal.txt](https://github.com/robotology/icub-support/files/1398865/yarprobotinterface_terminal.txt)

marcoaccame, 2017-10-19:
Hi @gabrielenava, I can see errors in the initialization of the IMU. And also moderate packet losses beyond the torso (torso-eb5, right_arm-eb3, right_leg-eb8, left_leg-eb6, right_arm-eb27) which cause failure of check-motion-done.

My guess is that there are electrical problems.

```
[ERROR]Bosch BNO055 IMU - Received unknown response message. 
         If this error happens more than once in a row, please check serial communication is working 
          fine and it isn't affected by electrical disturbance.
[ERROR]Bosch BNO055 IMU - Inertial sensor didn't understand the command. 
         If this error happens more than once in a row, please check serial communication is working 
          fine and it isn't affected by electrical disturbance.
[ERROR]Bosch BNO055 IMU - Received unknown response message. 
         If this error happens more than once in a row, please check serial communication is working 
           fine and it isn't affected by electrical disturbance.
[ERROR]FAILED writing  Set config NDOF_MODE 
[ERROR]Bosch BNO055 IMU - Inertial sensor didn't understand the command. 
         If this error happens more than once in a row, please check serial communication is working 
           fine and it isn't affected by electrical disturbance.
[ERROR]FAILED writing  Set config NDOF_MODE 
[ERROR]Bosch BNO055 IMU - Inertial sensor didn't understand the command. 
         If this error happens more than once in a row, please check serial communication is working 
           fine and it isn't affected by electrical disturbance.
[ERROR]FAILED writing  Set config NDOF_MODE 
[ERROR]Bosch BNO055 IMU - Inertial sensor didn't understand the command. 
         If this error happens more than once in a row, please check serial communication is working 
           fine and it isn't affected by electrical disturbance.
[ERROR]FAILED writing  Set config NDOF_MODE 
[ERROR]BoschIMU: set config NDOF_MODE failed 
```

```
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD 
torso-eb5-j0_2 IP 10.0.1.5 for nv ID32 = 0x0100020e -> IND = 2, TAG = 
eoprot_tag_mc_joint_status_core_modes_ismotiondone 

[ERROR]Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j=2) for
 BOARD torso-eb5-j0_2 IP 10.0.1.5

[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 37s 733m 308u: 
(code 0x00000030, par16 0x0002 par64 0x0000000000000021) -> SYS: the board has detected a 
hole in rx ropframe sequence number. In par64 there is the expected number. The received is 
obtained by adding par64 with par16 treated as a int16_t + .
```

julijenv, 2017-10-20:
Hi @marcoaccame , glad to hear that it is an electrical probelm but I was wondering why the firmware couldnt actually locate more specifically the source of it.

julijenv, 2017-10-20:
Hi all,
After rerunning the yarprobotinterface, I took the first error message as potentially the main error, after digging I couldnt really make my hand on something very shocking that could lead to some errors, I did recrimp one cable which was not perfect as the arm was popened it was now or never.
Anyway, afterwards I couldnt get the robot to get error messages so I checked the overall movements of the robot and did have to change cable N8 as well as retensed most of the cable of the shoulders.
wait for more feedback from you guys,
 

>

marcoaccame, 2017-10-20:
Hi @julijenv, 
the ETH board can always detect anomalies in ETH link as soon as it bootstraps, but it is robotinterface which does not listen to its messages until it has performed an handshake with it.

What happens in the green robot is the following. 

In `icub_all.xml` its is started in order:
- wrapper `head-mc_wrapper` which has `head-eb20-j0_1-mc` and `head-eb21-j2_5-mc`, 
- wrapper `torso-mc_wrapper` which has `torso-eb5-j0_2-mc`,
- wrapper `left_arm-mc_wrapper` which has `left_arm-eb1-j0_3-mc` etc.  

That means that the robot handshakes at first with EB20 and with EB21. These boards do not signal any problem, hence I am sure that the link
out of EB21 os OK.

After that, robotinterface handshakes with EB5 (but not yet with EB1/24/25/26). The board EB5 tells that there is a packet loss. 
Hnce, as I wrote, the problem is in teh left arm.

Can we tell where in a more precise way?

Try moving the start of LEFT ARM as the first wrapper. You can collect some more information.
You can also try changing the initialisation order of the EB1/24/25/26 boards to search for
the very link with problems.

```
<?xml version="1.0" encoding="UTF-8" ?>
<!DOCTYPE robot PUBLIC "-//YARP//DTD robotInterface 1.0//EN" "http://www.icub.org/DTD/robotInterfaceV1.0.dtd">

    <robot name="iCubGenova04" build="1" portprefix="icub">
    
    <params file="hardware/electronics/pc104.xml" />
 
<!-- remove it to debug electrical problems in left arm 
    <!-- HEAD -->
    <devices file="./hardware/motorControl/head-eb20-j0_1-mc.xml" />  
    <devices file="./hardware/motorControl/head-eb21-j2_5-mc.xml" /> 
    <devices file="./wrappers/motorControl/head-mc_wrapper.xml" /> 
 
    <!-- TORSO --> 
    <devices file="hardware/motorControl/torso-eb5-j0_2-mc.xml" />
    <devices file="wrappers/motorControl/torso-mc_wrapper.xml" /> 
  -->
  
    <!-- LEFT ARM -->
    <devices file="hardware/motorControl/left_arm-eb1-j0_3-mc.xml" />
    <devices file="hardware/motorControl/left_arm-eb24-j4_7-mc.xml" />
    <devices file="hardware/motorControl/left_arm-eb25-j8_11-mc.xml" />
    <devices file="hardware/motorControl/left_arm-eb26-j12_15-mc.xml" />
    <devices file="wrappers/motorControl/left_arm-mc_wrapper.xml" /> 
    
```

gabrielenava, 2017-10-23:
An update: with the help of @valegagge, we verified that the loss of communication between boards seems to be located between board 24 and 25 (left arm). There is probably something broken in between: can you  please @spada1978 help us to check exactly where is the source of this failure?

cc @julijenv @marcoaccame

spada1978, 2017-10-24:
il cavo giallo di alimentazione della potenza della scheda 26 ponticellato alla scheda 25 si è sfilato dal crimp quindi si è reso necessario effettuare nuovamente il crimpaggio 
Il lavoro è stato svolto insieme la collega Davide Gandini

julijenv, 2017-11-10:
@gabrielenava ,
can u double check if it has been resolved???

gabrielenava, 2017-11-10:
No other communication errors appeared during the last week while working on the robot, I think we can close the issue.

closed 2017-11-15 by julijenv