robotology/mesh-support #561: Fault button of  iCubGenova04 is broken

gabrielenava, 2017-11-06:
In particular, it seems that just touching the box (without pressing the button) is enough for inducing the fault. It happened twice that the robot was faulted because someone was just touching the fault box.

DanielePucci, 2017-11-27:
Any news on this? @julijenv @spada1978

julijenv, 2017-11-27:
Sorry guys i didnt see this one , I thought it was a double of the fault problem related to problem of visualization on yarpmotrgui. I'l have a look at it from thursday

julijenv, 2018-04-18:
Hi @DanielePucci , it has been already checked and I could not find any problem at all on the push button. So I asked @gabrielenava and @S-Dafarra , if they re-experienced the problem but noone answers me. So the real question is : Does it happen again?!?!? if yes please try to give more detail or try to be more specific on the occasion, if not just close the issue, thx in advance

S-Dafarra, 2018-04-18:
> So I asked @gabrielenava and @S-Dafarra , if they re-experienced the problem but noone answers me. 

Sorry @julijenv there should have been a misunderstanding. It is happening almost every time the fault button falls on the ground from a distance of more or less of 10cm or if someone place it on the desk a bit energetically (clearly not smashing it into the desk, but just a little bump is enough).

It does not seem to be happening by just touching it, but it is very sensible!

S-Dafarra, 2018-04-18:
Thanks @julijenv 

> btw You wanted to say sensitive I guess, not sensible

Well, maybe the fault feels bad when bumping :wink:

fiorisi, 2018-04-23:
This morning we noticed that the cable is damaged. See the picture attached. 
If possible, probably would be nice to change the cable with a thicker one.

[image]

julijenv, 2018-04-23:
Hi @gabrielenava , @S-Dafarra  and @fiorisi ,
please and get me ASAP the demo is over to let me start fixing this issue

julijenv, 2018-04-27:
Hi @gabrielenava @DanielePucci @S-Dafarra ,
it has been chasnged the averall cable, we checked with @S-Dafarra and it looks ok.

IITCODE Qta Description
189 x 3m : tehrmoresistrangete
2951 x 3m: cavo 4 poli

closed 2018-04-27 by julijenv

DanielePucci, 2018-04-27:
Thanks a lot @julijenv