robotology/mesh-support #564: icub-Genova03 [Skin sensors]

gonzalezJohnas, 2017-11-13:
### Description of the failure
The tactile sensors of the fingers of the right hand are not working

### Detailed conditions and logs of the failure
None

julijenv, 2017-11-13:
Hi @gonzalezJohnas ,
please open the drso cover and check if you can find one or more broken wire/s, or check at the level of the finger tip if one cable or more got ripped.
let me know

reafrancesco, 2017-11-14:
Hi @julijenv 
Sorry I couldn't check what you asked me because we have another issue, more critical :
- Right shoulder cable 7 is out of alignment and the shoulder pad can't be removed because since one of the four screws is stuck.

[image]
[image]
[image]

julijenv, 2017-11-15:
Hi @reafrancesco , You may request to EDL that they drill our the head of the screw in order to remove the external pulley to proceed with the substitution of cable 7. Do u have spare part for it???
Let me know.
PS: when drilling please use a vacuum cleaner in order to prevent aluminium bit getting into somewhere insidethe robot.
thx in advance

julijenv, 2018-01-16:
could anyone tell me if everything has been solved one your robot? @reafrancesco @gonzalezJohnas

gonzalezJohnas, 2018-01-16:
Hello @julijenv ,
Sorry for the delay ....
The robot right finger tip sensor still don't work, I didn't had time to check the dorso cover to check wiring cable issue.

About the rest of the issues everythings was done with you, except for the cover skin part we apply double tape.

julijenv, 2018-04-09:
Hi @gonzalezJohnas ,
please confirm if the last time I passed by to setup your robot there were still some fingertips not working and please just list the one not working (or that I said they were dead). If all working lease close the issue

gonzalezJohnas, 2018-04-11:
Hello @julijenv,

Sorry for the delay, the last time you came there were still fingertips not working ( you said that the taxel were probably dead).

Here are the part of the skins were patches are not responding on the skinGUI :
- Right hand (Index finger not working)
- Left and right forearms (Some patches are not responding, same in two forearms)
- Left and right arms (Some patches are not responding)

If you want more information I can try to record a video of the skinGUI to show you the patches inactive.

julijenv, 2018-04-11:
Hi @gonzalezJohnas and @reafrancesco ,
just remind you that you need to ask to Larosa for having (a) brand new patch(es) - the one glued on the skin- mainly because it takes time to get it done because we will need to take the part of the cover on which the patch is not working entirely and send it to a third company to glue the patch to it..
for the fingertip, just need to take a look at the ware house and I will bring it to you next time I come for a repair. Just remind me it, next time.
cheers,

closed 2019-02-13 by amaroyo