robotology/mesh-support #566: ethernet cable pc104

vtikha, 2017-11-16:
### Description of the failure
We have been having a large amount of problems with `iCubGenova01` where the robot would go suddenly in idle, frames would stop streaming or stream at 10fps. The problem lies with the connection with the 'pc104' where the system loses communication with it, and in the case where it gets restored the performance of the connection bandwidth would drop from **Gigabit ethernet** (1 Gbit/s) to **fast ethernet** (100Mbit/s) or even **ethernet** 10 Mbit/s. The only **solution** would be to restart the `pc104` (or `ifdown` & `ifup`) but it could be that more than one restart (or reset of the network interface) is necessary to return to good working order. 

With @mbrunettini we tried connecting directly to the pc104 and were not able to replicate this issue, but, as it is not deterministic, we can only state this. We also put `iCubGenova01` on a separate switch to cut out eventual hardware problem which proved not to be the problem. 

As this is a **critical issue**, as at this stage`iCubGenova01` is **unusable**, and since we previously tried fixing this problem by adjusting the connectors directly on the `pc104` without any success, we propose replacing the whole ethernet cable.

people involved in tests:
@pattacini @randaz81 @mbrunettini

julijenv, 2017-11-22:
Hi @vtikha ,
The ethenret cable ahas been changed coul you please run some tests to make sure the issue is solved?!?!
thx in advnace

vtikha, 2017-11-22:
@julijenv great. We have started making various tests and I can report that, so far, the above problems have not occurred. Thank you for your swift support, let's close this issue for now.

closed 2017-11-22 by vtikha

pattacini, 2017-11-22:
cc @claudiofantacci @giuliavezzani