robotology/mesh-support #567: Right wrist problem

gsaponaro, 2017-11-20:
Hello, we think that we have an encoder problem, due to the following symptoms:

1. During robot calibration, the joints of the right wrist (j5 r_wrist_pitch & j6 r_wrist_yaw) go to the wrong physical position. This is a video with the problem visible at the end:
https://youtu.be/WiGfogvUAPQ

Here is the **yarprobotinterface** log: https://gist.github.com/gsaponaro/266739ae5a0a9e2ea8cf3cfd53b18e05

At the end of the calibration, with the wrist in the wrong position, **yarpmotorgui** says that j5 and j6 are both at zero.

2. Additionally, we notice that the encoder values are **drifting** with time. Initially (after calibration) they look like this:

[image]

Then, we idle the joints and we move the wrist up and down a few times, and they look like this:

[image]

**Things that we already checked so far, but did not solve the problem**:

We checked the wires continuity of the relevant CFW2CAN2 board with label 2B2, id 3.

We checked the encoder output with an oscilloscope, it looks OK (as far as we know) because we see two square waveforms that trigger when we move the wrist in idle.

We re-flashed the firmware 4DC.1.19.out.S on the board.

We tried with a spare (brand new) control board that we had here.

Any suggestion?

/cc @rnun3s @vicentepedro

julijenv, 2017-11-22:
Hi @gsaponaro ,
 Could you please check if  the ribbon cables of both wrist motors arent a bit worn and touched some metallic part of the robot. Especially when in position or  about to be calibrated, it  appeared that at the end of the calibration the motor jump from good posiiton to bad one. If you relaunch the calibration try to detect which of the 2 wrist motor moved, it could help you to detect a problem in this motor and check for the reason. (In the worst case scenario - if you cant see any worn part, just recrimp the motor wire inside its connector going into the board 3B2.
Let me know

gsaponaro, 2017-11-22:
Hi @julijenv , thanks for the help. The wrist motor responsible for the problem is the lower one. We have already checked the continuity of its wires, plus the encoder output with an oscilloscope. We will look for you on Skype, OK? Cheers.

julijenv, 2017-11-22:
Hi @gsaponaro ,
the continuity test is not enough, from what I can see from the video, you may have a leak (check for worn part of the cable) - Check if the carcasse is 0V or higher? (from any black wire on the robot and the other probe on the carcasse - main part of the robot). if everything seems ok, just recrimp the 6 wires of the motors and let me know,

gsaponaro, 2017-12-07:
Hi @julijenv,
sorry for the delayed response. We have performed the checks:

* possible tension leak: carcasse is at zero, or very slightly above zero, probably not significant (0.02V)
* we have re-crimped the 6 wires of the wrist motors (both lower and upper motors)
* now, during calibration the top motor does not move at all, while the bottom one moves. Fresh video here: https://youtu.be/fisFOrDhGG4, yarprobotinterface log here:
https://gist.github.com/gsaponaro/726a6926fe83f62e5bcf7d150df51583
showing some right arm (CAN2) errors in lines: 1451, 1531-1532;
* we have also tried swapping the wrist boards among the two arms (i.e., swapping 1B2 with 2B2) - and the behaviour of the right wrist is same. Therefore, **the problem seems in one of the motors**.

gsaponaro, 2017-12-07:
Update: we applied some voltage to the terminals of the **upper motor**, directly to to the "wire exit" of the motor as you can see in this photo:

[image]

Result: the motor does not rotate.

julijenv, 2017-12-11:
Hi @gsaponaro ,
Seems like the motor is dead so.
So I guess you may need to ask for a replacement.
please get in touch with andrea.derito@iit.it to get a quotation.
thx

gsaponaro, 2017-12-11:
Thanks, just a remaining question:
should we separate the black motor part from the metal gearhead? (e.g., if it makes the replacement easier, or if you want the dead motor for some analysis.)

Pictures below.

[image]
[image]

julijenv, 2017-12-13:
Hi @gsaponaro,
Unfortunately the black part is not a substitutionnal piece, fromwhat I know, we buy the motor with the gear box all attach, never have to separate them. The only think that you may need to know is that first you should try to unmount the threading pulley (if you can by removing the pin) and afterwards the metallic piece with the 2 pulleys on it.
If you can do it, you will just need to replace the motor. if you can't, you will also need to buy the motor with all the pieces that you have in the pictures you took.
let me know

gsaponaro, 2017-12-13:
Thank you. We managed to dismount the metallic piece w/pulleys part.

We will send the email order specifying: wrist motor including its gearbox, with the hole already made on the shaft, but with no metallic piece w/pulleys.

gsaponaro, 2018-02-07:
Hello, we received and installed the new motor (replacing the upper motor which was dead).

Unfortunately, the problem remains. As in the initial post of this issue, we observe two problems: (1) calibration of j5 r_wrist_pitch & j6 r_wrist_yaw go to the wrong physical position as in the video https://youtu.be/WiGfogvUAPQ; (2) when we idle the joints and we move the wrist up and down a few times, the encoder values drift to very high numbers.

Our feeling is that there is a further problem on the control board, or in the encoder of the lower motor.
Any suggestion of how to debug these?
Another question: are there any encoders contributing to the joint values j5 & j6, in addition to the two encoders attached to the two motors?

Thank you.

julijenv, 2018-02-08:
Hi @gsaponaro ,
The first thing that comes into my mind is to know if you correctly crimped the 6 motor cables and if you connected them correctly, You should follow the same scheme of wiring as the colour cord of the motor. To be more clear try to copy the wiring of another motor with similar colour cord - there 2 types of power cord for the motor inside the upper arm - a grey blueish and and a grey yellowish, they are wired differently. 
If these 2 checks are done (crimp and wiring) I would suggest you to try to swap both boards as you did as a pretest from left arm and right arm and see if the motor moves correctly.
Last test but means that the brand new motor has been incorrectly mounted with enough care (we perfromed a test on every motor bougt and we also test it before it has been sent) and has been damaged (or during the shipping who knows) is making a continuity test in between the crimp and one-two centimeters away from connector and why not also uses a power supply to put some volts inside the motor to try to see if it moves.
let us know

gsaponaro, 2018-02-09:
We have some good news and some bad news.

Good: we confirmed that the new spare motor (upper wrist motor) that you sent us is OK.

Bad: unfortunately it seems that the **encoder of the lower wrist motor** is broken and so that part needs to be replaced, too. We say this after having performed the following tests:
* we replaced the 2B2 board with a new spare board (verified working in the other arm 1B2)
* we re-crimped the wires, and confirmed that continuity of the motor wires is OK 
* we tested the encoder of this lower motor (with an external 4DC control board)
* moving the wrist of the robot up and down a few times, the encoder values drift to very large numbers
* the lower motor rotates OK (we applied a voltage and it runs), the problem is only in the encoder, possibly a broken tooth inside

Note: currently some of our members are at VVV18. Would it be possible to give them the spare before they travel back on the 15th? Thanks, let us know if you need more information or tests.

gsaponaro, 2018-02-09:
Sidenote - perhaps the following information is useful for somebody, or for reducing costs. Regarding this point:

> Unfortunately the black part is not a substitutionnal piece, fromwhat I know, we buy the motor with the gear box all attach, never have to separate them. 

recently we did manage to separate the black motor part from the metal gearhead with some hot air:

[image]

julijenv, 2018-02-12:
Hi @gsaponaro ,
could you be more clear about which part you would like to get a spare, you mean the motor?!?!?
Unofortunately, there is no way I could give you a spare like that without having done an order for it.
Could you try to get in touch with me on skype, the earlier the better to understand the exact spare require and see what we can do to accelerate teh process to perhaps give it to your team member in VVV18.

vicentepedro, 2018-02-12:
Hi @julijenv,

Only the **encoder** is broken. The motor can rotate when we apply some voltage.
However, we are not sure if it's possible to replace just the **encoder**, or if we need a **new motor** instead...

Can you tell me your skype name? Probably, I can explain it better on a skype call.

Best,
Pedro

julijenv, 2018-02-12:
Hi @vicentepedro and @gsaponaro ,
Unfortunately it is the same problem as the previous one, the motor comes like it is to us(encoder+motor+gearbox) we cant buy them independently, the other bad notice if the encoder is broken now on the lower wrist motor is to not only send you the motor but we will need to drill 2 holes (1 for the letting the cable passes and the second one to block the coupling in between the threaded pulley and the main shaft of the motor).
I would advise you to get in touch with Andrea D. for the spare part and if can send back the same parts thta you did before it would be better or else you will have to do it by yourself.
let me now

julijenv, 2018-03-23:
Hi @gsaponaro ,
I didnt get any news from you since some times, but I've just seen pass a couple of mails from andrea Derito who was asking about the pin from finger but nothing about the motor, Did u succeed to figure out that the motor was ok at the end?
let me know

gsaponaro, 2018-03-23:
Hi @julijenv ,
Sorry for the delay.

The new upper wrist motor that you sent us is OK, thanks (the emails with Andrea are about that one).

About the lower wrist motor (where only our encoder is broken): in the end, we are going to purchase just the encoder+motor part from another supplier, and then "recycle" our working gearbox that we have managed to separate ([picture](https://github.com/robotology/icub-support/issues/567#issuecomment-364504339)).

From the point of view of IIT you can consider this issue finished, feel free to close it now if you prefer.

From our point of view, the wrist is not yet working, hopefully it will be in a couple of weeks when we receive the spare part from the other supplier.

julijenv, 2018-03-23:
HI @gsaponaro ,
when you will have your robot working again, please close it or let me know , like that I will close it, cheers

gsaponaro, 2018-06-28:
This issue is now solved, thanks for the help!

> About the lower wrist motor (where only our encoder is broken): in the end, we are going to purchase just the encoder+motor part from another supplier, and then "recycle" our working gearbox that we [...]

For our personal record, we purchased the encoder+motor part from Moverica / Elmeq in Spain.

closed 2018-06-28 by gsaponaro