robotology/mesh-support #568: Need of Longer power supply and communication Cable for iCub

MichaelBombile, 2017-11-22:
Hi @DanielePucci , in your video of iCub balancing (https://www.youtube.com/watch?v=VrPBSSQEr3A), you have a longer power supply cable. It is possible to get such a long cable for iCubLausanne02?
If possible, what is the procedure that we must follow to get one?
Thanks in advance.

CC: @asutcl

julijenv, 2017-11-22:
HI @MichaelBombile ,
could you please send me a picture of the connector coupling your power supply to the power supply cord? afterwards I will put you in contact with the purchase office.
thx in advance

MichaelBombile, 2017-11-22:
Hi @julijenv ,
Thanks for your response. In fact, there is no particular connector between the power supply and the power supply cord. Here are some pictures of their connections.

[image]
[image]

And on the robot side
[image]

Thanks.

julijenv, 2017-11-22:
Hi @MichaelBombile ,
Ok I know What you need now.
Do you mind to give me the contacts of yours and your supervisor in order to send you a quotation for this 7m long extension cable.
Please send me a mail with the request at julien.jenvrin@iit.it
thx in advance

closed 2017-11-22 by julijenv