robotology/mesh-support #570: Boards communication problem

gabrielenava, 2017-11-23:
It happened again (see https://github.com/robotology/icub-support/issues/555). In particular, while the `yarpRobotInterface` is running, the left lower arm starting from the `wrist pronosup` joint is not responding anymore. This happens randomly both with the robot torque controlled and position controlled. Here is the logger for two different episodes: 

[dataLogger_2.txt](https://github.com/robotology/icub-support/files/1499736/dataLogger_2.txt)
[dataLogger.txt](https://github.com/robotology/icub-support/files/1499737/dataLogger.txt)

next time it happens, I will make use of `FirmwareUpdater` to locate more precisely which board is not responding, so we know where to look. Most likely, it is a broken cable somewhere.

marcoaccame, 2017-11-24:
the firmware updater will not tell you anything because boards run smoothly. 
the problem is transient electrical noise or a eth cable which sometimes (with some movements) does not work well or a powers supply cable which sometimes (w/ some movements) touches the body.

from the log files, problems are in:
- connectors 10.0.1.1:P3 (or P2 of next board) and 10.0.1.3:P2 (or P3 of next board).
- continuous CRC errors detected in 10.0.1.25:P3

here is why:

first log file:

-   @ 796s 177m 186u the 10.0.1.1:P3 link goes down  
-   @ 808s 678m 187u the 10.0.1.1:P3 link goes up 
-   @ 808s 677m 131u the 10.0.1.3:P2 link goes up 
-   @ 811s 877m 195u the 10.0.1.1:P3 link goes down
-   @ 814s 78m 186u the 10.0.1.1:P3 link goes up
-   @ 814s 77m 131u the 10.0.1.3:P2 link goes up

second log file:

-   @ 381s 100m 196u teh 10.0.1.25:P3 detects CRC errors
-   @ 381s 500m 196u ....
-   etc. for about 20 times + packet loss everywhere
-   @ 501s 957m 711u the 10.0.1.1:P3 link goes down 
-   @ 504s 58m 676u the the 10.0.1.1:P3 link goes up

DanielePucci, 2017-11-27:
Any news on this? @julijenv @spada1978

DanielePucci, 2017-11-27:
Right now the hardware fault is blocking the normal activities. We really need support on this @julijenv @marcoaccame @spada1978

DanielePucci, 2017-11-28:
This problem keeps occurring @maggia80

marcoaccame, 2017-11-28:
Hi @DanielePucci, we (@maggia80, andrea mura and I) looked at the robot at about lunchtime. We are aware of the situation and we are looking at it more closely. As far as I know, however, since after lunchtime it did not happen again.

DanielePucci, 2017-11-28:
Thanks @marcoaccame for taking care of this issue. I do not know the exact timing of today's experiments, but I know that today it happened twice. Even if it did not happen after lunch, this issue must be carefully looked at (as you are kindly doing :) )

maggia80, 2017-11-30:
Mura, Emiliano and me with the help of guys from Pucci's lab, we debug quite a lot.
The issue was generated by a voltage drop in the MC4-PLUS board of the upperarms. In particular, the board EB24 was the one that was rebooting. 
The voltage drop (9.7V instead of 12V with all the 12 motors controlled but in position and the undervoltage of the MC4-PLUS is 7.8V) was caused by a problem in the wiring. We fix the wiring, with a modification in order to make it more stable and reliable and we also add a capacitor (680uF) for removing the extra drop when the forearm and hand motors move. We did it for both arms. Now the voltage is closed to 12V.

We experienced also an overcurrent fault on the 12V  DC/DC Regulator of the BDC (the board in the backpack). We increase the current limit to 10A instead of 6A by changing the FW. 
note: once programmed the board BDC we left for a while the programming cable attached to the BDC, however, it was causing a reboot of the microcontroller of the BDC , especially when someone was touching the robot after being electrically charged on the chair.  One removed the cable the issue disappeared. I have tested it two times only since the spikes you get every time is quite strong, but please Dani go ahead with the testing :-) 

I will close the issue but please add comments and reopen it if you get the same issue.

closed 2017-11-30 by maggia80