robotology/mesh-support #571: Right leg FT sensor may be broken

gabrielenava, 2017-11-23:
In particular, during the last experiment on iCubGenova04, we restarted the robot after a failure. After calibrating the robot in the air, this were the forces seen by FT sensors:

[image]
 
the robot is on two feet and no-one is touching it, so this estimation is completely wrong. Furthermore, we checked the output of FT sensors (ports names `icub/....../analog:o`) and the output of the right leg FT was zero (I was touching the robot to see how the FT were responding):

[image]

restarting again the robot didn't change anything. So I think the FT is broken, or maybe a cable got disconnected.

fiorisi, 2017-11-24:
The new FT has `half-gain` configuration, is this correct?

fiorisi, 2017-11-27:
No, the new FT has the new electronic board `strain 2` with adjustable gains.

julijenv, 2017-12-01:
Could you confirm. Is it still an open issue??? @fiorisi @gabrielenava

fiorisi, 2017-12-05:
@julijenv thank you, the FT has been replaced.

closed 2017-12-05 by fiorisi