robotology/mesh-support #572: iCubErzelli01: Failure on right arm EB27

ddetommaso, 2017-11-24:
### Description of the failure

A failure on the right arm again on iCubErzelli01.
The right arm does not calibrate properly during the yarprobotinterface.

The logfile shows at first the following errors

736 14,987290 <ERROR> embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD right_arm-eb27-j4_7 IP 10.0.1.27 for nv ID32 = 0x0100000c -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_controlmodestatus 
 
737 14,987414 <ERROR> Failure of askRemoteValue() inside embObjMotionControl::doneRaw(axis=0) for BOARD right_arm-eb27-j4_7 IP 10.0.1.27

### Detailed conditions and logs of the failure

Running the yarpmotorgui the following joints result not properly configured: J4, J5, J6 and J7.
[image]

Running the ethLoader the EB27 responds correctly and also the ping works.
[image] 

Please attached you can find the complete logfile.

[log_icub-head_yarprobotinterface_3223.txt](https://github.com/robotology/icub-support/files/1501840/log_icub-head_yarprobotinterface_3223.txt)

julijenv, 2017-11-27:
Hi @ddetommaso ,
AS you mentionned it is one more time the board 27 which has an issue, but this time looks like it is different from last itme (where i had to changed it as we couldnt get in touch with it anymore). This time it looks like the board is repsonding somehow even if it doesnt make the calibration.
Could you please try to relaunch the robbot without/with shutting down the motor and see if it restarts sometimes and getting the how sporeadics it is or if it is repeatable.
If you succeed to get it calibrted could you try to make it move (arm adn forearm) but minly the 4 motors attached to it from pronosuppination to finger abduction and see if the board got disconnected or if it goe sin hardware fault for some specific movement.
let me know,
ill be back on thursday i'll pass by erzelli before the end of the week

julijenv, 2017-11-27:
Hi @valegagge and @marcoaccame , could you have a look at the log file of this issue and give me more inside of what could have been happening, let me know - The strange thing is that it looks like it is always the boards from the forearms joints that have this issue?
Did you have a wayt o test if it is a drop in power line or if it is a network drop only?
let me know

marcoaccame, 2017-11-28:
hi @julijenv, yesterday afternoon i talked directly to fabrizio la rosa. i did not have tiem to reply on github.

here is what i wrote:

```
Vari problemi:
1. Alcuni Aea hanno spikes... controllare 
2. Link eth va giù sulla eb27. Cavi da controllare 
```

marcoaccame, 2017-11-28:
Hi Julien,

 ***the board EB27 had a reset and restarted***

here are some more details:

following messages tell that the outbound ETH link of board EB3 on connector P3 went down and it stayed down for 2 seconds 101 milli. there was no reset of the EB3 board because application state is running (most significant nibble of par64 is 0x3).

```
from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 10495s 166m 835u: 
(code 0x06000001, par16 0x0001 par64 0x3000000000000000) -> ETH monitor: link goes down
in port specified by par16 (0 = P2, 1 = P3, 2 = internal).  Application state is most 
significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .

from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 10497s 267m 800u: 
(code 0x06000000, par16 0x0001 par64 0x3000000000000000) -> ETH monitor: link goes up 
in port specified by par16 (0 = P2, 1 = P3, 2 = internal). Application state is in most 
significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .

```
About at the same time, board EB28 signals that its incoming link is up again. this board did not have a reset.

```
from BOARD 10.0.1.28 (right_arm-eb28-j8_11), src LOCAL, adr 0, time 10497s 175m 306u: 
(code 0x06000000, par16 0x0000 par64 0x3000000000000000) -> ETH monitor: link goes up
 in port specified by par16 (0 = P2, 1 = P3, 2 = internal). Application state is in most 
significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. + .
 ```

So far no information about what happened to board EB27 in between the above two.

But just after a while ...

```
829 22,499327 <ERROR> hostTransceiver()::onMsgReception() detected an ERROR in sequence 
number from IP = 10.0.1.27. Expected: 6344, Received: 1, Missing: 18446744073709545273, 
Prev Frame TX at 7900821091 us, This Frame TX at 3101612 us 
 .....
831 22,499496 <DEBUG> EthMonitorPresence: BOARD 10.0.1.27 (right_arm-eb27-j4_7) has shown 
after being lost for 8.73341 sec 
```

This is important information: 
**It tells that the board EB27 had a reset and restarted.**
Because:  a board transmits with sequence number 1 only if it is its second UDP packets sent out. And that is possible only if the board just bootstrapped into the application. Moreover: the service EthMonitorPresence inside pc104 sees again the board after 8.73341 sec, which is a time compatible with a restart (5 seconds in updater process, plus a few seconds in the application to startup sockets).

maggia80, 2017-11-30:
The issue is similar to the #570. However this case there was also a MC4-PLUS card with a strange behavior. We are going to test this card in order to figure out which is the isssue.
Emiliano and @davidetome emoved the faulty card and install a new one.
Moreover, they add a 680uF capacitor in both hands.

closed 2017-11-30 by maggia80