robotology/mesh-support #577: The iCub laptop doesn't turn on

dhgarcia, 2017-11-29:
### Description of the failure

The iCub's laptop (icubsrv) is dead. I doesn't turn on when powered.

### Detailed conditions and logs of the failure

The iCub doesn't turn on. Nothing happens when pressing the ON button. It was working fine until yesterday.
We have try unplugging the AC adaptor, moving to a different location. Removing the battery. 
It is still dead.
The laptop is a DELL E5550

No one left here in Plymouth was around when all things were setup. 
Did IIT provided the laptop or we need to find a new one?
How do we configure a new laptop to act as the icub's laptop server?
Is there any way to use the icub without the laptop?

Thanks for your help.

julijenv, 2017-11-30:
Hi @dhgarcia ,
to answer to question one after another:
1- Did IIT provided the laptop or we need to find a new one?
Yes IIT provided you the laptop that it is from what I can read from you - DEAD. I think the best way to proceed with iCub is to buy another laptop OR use another working station as server.
2- How do we configure a new laptop to act as the icub's laptop server?
just follow the instruction from the wiki page - http://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions 
3 - Is there any way to use the icub without the laptop?
of course you can use another machine (laptop or workstation) and set everything as specify from the link on answer from question 2.
Let us know,

dhgarcia, 2017-11-30:
Thank you @julijenv 
We will get a new laptop and configure it following the wiki.
Is there any special specification (recomendation) we should have in mind for the laptop to buy?

best,
Daniel

julijenv, 2017-11-30:
@mbrunettini , could you advise them a minimal requisite?!!? thx in advance

mbrunettini, 2017-11-30:
Of course it depends on your needs, there no "minimal".
We currently use DELL Precision 7520 with 

- 32 Gb of RAM
- 512 SSD
- Intel i7 processor
- nVidia discrete video card.

BarisSerhan, 2018-01-03:
Hi @julijenv,

We bought a new laptop and I followed all the installation instructions for icub laptop server. I first installed Ubuntu 16.04 however it seemed like this distribution is too recent to work compatibly with the one of pc104 (version 5.0-15.11.02 Debian wheezy), they had different versions of some of the libraries. Then I downgraded the OS to Ubuntu 14.04 and followed all the instructions again but I cannot still run yarp on the pc104 . 

When I run the "yarprun" command, I get this error message:

`yarprun: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.19' not found (required by /usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1)
yarprun: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.14' not found (required by /usr/local/src/robot/yarp/build/lib/libYARP_OS.so.1)`

Do you have any idea to fix the problem? (might it be related to gcc versions? icub laptop has 4.8.4 while pc104 has 4.7.2)

Our situation is urgent since tomorrow there will be an important photo shoot with icub for a cover of a magazine, any help would be greatly appreciated.

Kind regards,
baris

julijenv, 2018-01-04:
Hi @BarisSerhan ,
First thing first, you are using iCubPlymouth02 - so it means it is just a head nothing else, Am I wrong?

Now, type on the pc104 before changing anything,

> uname -a 

and see if the pc104 is compatible 64bits. and also please send me a picture of the cfw (the can board on which you should have connected the can0 - at the level of the neck behind the head. If it is a cfw2 you are good to proceed.
Why upgraded to the most recent? the reason is simple, using the most advance features develop by the icub community.

Now I would suggest you to retrieve one more time the 16.04 version for the laptop - server- becasue the 14.04 is not anymore supported. outdated source files. I would suggest you to get a new usb key (8GB or more) and fetch the new release or the one that could fit the version of the head you've got.
FOr that try to take a look at the wiki page of iCub:
http://wiki.icub.org/wiki/The_Linux_on_the_pc104

Let us know

pattacini, 2018-01-04:
@julijenv I and @BarisSerhan talked each other via email yesterday and I pointed out a thing that wasn't probably very clear: the need for compiling/running code meant for PC104 directly from the PC104 via `ssh`.

Instead, running code meant for PC104 that has been compiled on the server will bring about incompatibility issues.

julijenv, 2018-01-04:
Hi @BarisSerhan ,~
a question remains, does that issue cleared? Is it solved?
let us know

BarisSerhan, 2018-01-05:
Hi @julijenv and @pattacini,

The robot is the iCubPlymouth01 so it is the whole robot and 64bit compatible. Many thanks again to @pattacini for helping me to sort out some fundamentals. I compiled everything according to the instructions from pc104. Even if I could managed to run  `yarprun` and received input from icub’s camera, `yarprobotinterface` threw some errors. I fixed some of the errors by changing configuration files but there were also some errors related to canmotion that I couldn’t sort it out. Since we had the photo shoot yesterday, I had needed a quick solution so I unscrewed the old laptop, took its hard disk out and copy the content of the export folder to the new laptop. Hopefully icub started to work again and everything went well yesterday.

So for now the issue is solved and it can be closed, however it would be nice to update pc104 and its icub software repositories in near future since we are running the old versions. 

Thanks for your supports.

Regards,
baris

closed 2018-05-09 by dhgarcia