robotology/mesh-support #581: Lower left leg skin doen not starts

gabrielenava, 2017-12-05:
When turning on the motors, the left lower leg skin does not start (there are no blue lights on). Looking inside the leg without removing the cover, it seems every cable is connected. I attached to the issue the logger messages when starting the robot.

[log_leftLegSkinNotWorking.txt](https://github.com/robotology/icub-support/files/1531806/log_leftLegSkinNotWorking.txt)

julijenv, 2017-12-11:
looks like the CAN2 of the EMS doesnt give enough power. EMS cannel CAN2 probably dead.

julijenv, 2018-06-25:
This board has been replaced by :
 4546.F x 1 : EMS4, IIT - Electronic board, Ethernet Motor Supervisor (EMS) board, with Cortex-M4 processor (With STM32F407 Rev.

closed 2018-06-25 by julijenv