robotology/mesh-support #583: Cannot compile icub-main on pc104 with latest releases

EliseiFrederic, 2017-12-07:
### Description of the failure
icub-main compilation fails when building cfw2can, with libcfw002.h not found

### Detailed conditions and logs of the failure

On a fresh install (new icubsrv laptop), with the last Debian image for pc104,
http://www.icub.org/download/pc104-images/icub-live_7.9-17.12.04-rt-amd64.zip
and the latest sources from git.
Firmware was not updated (should it be ?), all other things were built/configured freshly.

Scanning dependencies of target cfw2can
[ 23%] Building CXX object src/libraries/icubmod/cfw2Can/CMakeFiles/cfw2can.dir/yarp_plugin_icubmod_cfw2can.cpp.o
In file included from /usr/local/src/robot/icub-main/build-pc104/src/libraries/icubmod/cfw2Can/yarp_plugin_icubmod_cfw2can.cpp:10:0:
/usr/local/src/robot/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.h:38:23: fatal error: libcfw002.h: No such file or directory
 #include "libcfw002.h"

kt10aan, 2017-12-08:
It may sound trivial (but it has happened in the past) so it's worth a try.
. Are you sure you have sshed in the pc104 and ccmaking/compiling from there?

If you have, during ``ccmake``, check if the ``cfw2`` headers/libraries are found.

EliseiFrederic, 2017-12-08:
Thanks for the suggestion, but I was on PC 104 (much slower compilation with the atom processor !) and in a dedicated build directory

The .so is found automatically by ccmake (it is a realtime module, directly included in the live image). The header file is not ! But it stays at the same location in the live
+image. I did try to include this directly. That's enough for cfw2can to compile. But then, yarprobotinterface fails to load cfw2can.
Same for Firmware update : cfw2can is found but fails to load.

The can hardware is fine : with the previous laptop and USB key (décembre 2015) everything is working

EliseiFrederic, 2017-12-08:
If I change in /icub-main/src/libraries/icubmod/cfw2Can
the #inclusion of 
#include "/lib/modules/4.9.0-3-rt-amd64/iCubDrivers/cfw002/LinuxDriver/API/libcfw002.h"
instead of
//#include "libcfw002.h"

I can compile the full icub-main suite. But yarprobotinterface still fails during startup : 

[log_new_yarprobotinterface.txt](https://github.com/robotology/icub-support/files/1542626/log_new_yarprobotinterface.txt)

barbalberto, 2017-12-12:
Can you the following, please?
Remove the build folder for icub-main on pc104
run `yarpdev --list` and make sure there is no canmotioncontrol module
run `ccmake` turning on canmotioncontrol flag
then run `cmake` (single 'c') from terminal and post the output.
compile everything and run 
`yarp plugin --verbose canmotioncontrol` and post the output

EliseiFrederic, 2017-12-12:
Thanks for your help. I think I did the first steps correctly... 

Le 12-12-2017 09:38, Alberto Cardellino a écrit : 

> Can you the following, please?
> Remove the build folder for icub-main on pc104
> run yarpdev --list and make sure there is no canmotioncontrol module
> run ccmake turning on canmotioncontrol flag
> then run cmake (single 'c') from terminal and post the output.

See attached file for this log. 

> compile everything and run

Do you mean, with no more flags than enabled canmotioncontrol, or with
all the usual flags ? 

> yarp plugin --verbose canmotioncontrol and post the output

This is from the new build-pc104 directory (with only canmotioncontrol
flags) 

icub@pc104:/usr/local/src/robot/icub-main/build-pc104$ yarp plugin
--verbose canmotioncontrol 
Yes, this is a YARP plugin
Trying to find library
'/usr/local/src/robot/yarp/build-pc104/lib/yarp/canmotioncontrol.so'
containing function 'canmotioncontrol' -- fail
error while opening
/usr/local/src/robot/yarp/build-pc104/lib/yarp/canmotioncontrol.so:
 /usr/local/src/robot/yarp/build-pc104/lib/yarp/canmotioncontrol.so:
cannot open shared object file: No such file or directory
Trying to find library
'/usr/local/src/robot/icub-main/build-pc104/lib/iCub/canmotioncontrol.so'
containing function 'canmotioncontrol' -- found
 * library:
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/canmotioncontrol.so
 * system version: 5
 * class name: yarp::dev::CanBusMotionControl
 * base class: yarp::dev::DeviceDriver

I would get what seems to be the same from my previous build-pc104
directory with all the flags : 

mv build-pc104 build-pc104_test
mv removed_build-pc104 build-pc104
cd build-pc104
make 

icub@pc104:/usr/local/src/robot/icub-main/build-pc104$ yarp plugin
--verbose canmotioncontrol 
Yes, this is a YARP plugin
Trying to find library
'/usr/local/src/robot/yarp/build-pc104/lib/yarp/canmotioncontrol.so'
containing function 'canmotioncontrol' -- fail
error while opening
/usr/local/src/robot/yarp/build-pc104/lib/yarp/canmotioncontrol.so:
 /usr/local/src/robot/yarp/build-pc104/lib/yarp/canmotioncontrol.so:
cannot open shared object file: No such file or directory
Trying to find library
'/usr/local/src/robot/icub-main/build-pc104/lib/iCub/canmotioncontrol.so'
containing function 'canmotioncontrol' -- found
 * library:
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/canmotioncontrol.so
 * system version: 5
 * class name: yarp::dev::CanBusMotionControl
 * base class: yarp::dev::DeviceDriver

Hope it helps, thanks ! 

[... 146 lines cut ...]
-- Installing cmake scripts
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/src/robot/icub-main/build-pc104

icub@pc104:/usr/local/src/robot/icub-main/build-pc104$ printenv YARP_DATA_DIRS
/usr/local/src/robot/yarp/build-pc104/share/yarp:/usr/local/src/robot/icub-main/build-pc104/share/iCub:/usr/local/src/robot/iCubContrib/share/ICUBcontrib:/usr/local/src/robot/icub-tests/suits

icub@pc104:/usr/local/src/robot/icub-main/build-pc104/yarp$ grep YARP_DATA_DIRS ~/.bashrc_iCub 
export YARP_DATA_DIRS=${YARP_DIR}/share/yarp:${ICUB_DIR}/share/iCub:${ICUBcontrib_DIR}/share/ICUBcontrib:${ROBOT_CODE}/icub-tests/suits

[ 25%] Built target icubmod
Scanning dependencies of target canmotioncontrol
[ 25%] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/yarp_plugin_icubmod_canmotioncontrol.cpp.o
[ 25%] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function ‘unsigned char yarp::dev::CanBusMotionControl::from_modevocab_to_modeint(int)’:
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3967:17: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         return  VOCAB_CM_UNKNOWN;
                 ^~~~~~~~~~~~~~~~
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3976:16: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         return VOCAB_CM_UNKNOWN;
                ^~~~~~~~~~~~~~~~
[ 26%] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/ThreadTable2.cpp.o
[ 26%] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/ThreadPool2.cpp.o
[ 26%] Linking CXX shared module ../../../../lib/iCub/canmotioncontrol.so

barbalberto, 2017-12-13:
> Do you mean, with no more flags than enabled canmotioncontrol, or with
all the usual flags ?

Sorry, I meant with all usual flags, but `canmotioncontrol` only is enough for the tests done.

When you compiled, the header was still modified or reset as the original? In it was modified, can you try to reset it and see if it compiles now? (if not we will have to check more, but it is a different matter from the robotinterface not starting, see later)

The plugin (when compiled) is correctly found by yarp. The yarprobotinterface is not starting because the log states that config files are old and no more compatible with latest version of `canbusmotioncontrol` device.

```
[ERROR]AxisName not found
[ERROR]AxisType not found
[ERROR]fullscalePWM not found
[ERROR]fullscalePWM param not found in config file. Please update robot configuration files or contact https://github.com/robotology/icub-support
[ERROR]One or more errors found while parsing device configuration 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
```

Also ControlBoardWrapper2 config files should be updated

```
[WARNING] ControlBoardWrapper2 device:
************************************************************************************
* ControlBoardWrapper2 'name' parameter for port name does not follow convention,  *
* it MUST start with a leading '/' since it is used as the full prefix port name   *
*     name:    full port prefix name with leading '/',  e.g.  /robotName/part/     *
* A temporary automatic fix will be done for you, but please fix your config file  *
************************************************************************************ 
```
This latter is already updated upstream in the `https://github.com/robotology/robots-configuration` repository so you just need to pull it.

For the `canmotioncontrol` parameters I think @randaz81 can help you out.

EliseiFrederic, 2017-12-13:
Thanks for the updated ControlBoardWrapper2 files !
Is it a clue that the firmware are likely to need update ?

cfw2can still cannot be compiled with the original #include "libcfw002.h".
This is the single module not compiling. No impact on canmotioncontrol.

EliseiFrederic, 2017-12-13:
icub@icubsrv:/usr/local/src/robot$ cd robots-configuration
icub@icubsrv:/usr/local/src/robot/robots-configuration$ git pull
Already up-to-date.

Am I missing something there ?

barbalberto, 2017-12-13:
Right now it is too early to determine if a firmware update is required.
Steps are:
- robot-configuration:  pull and `make install` to install new files and remove ControlBoardWrapper2 warning.
- updated `canbusmotioncontrol` config file with @randaz81 support and again `make install` to install new files.
- When `canbusmotioncontrol` starts, it'll check the firmware version and tell you if an update is required.

randaz81, 2017-12-14:
Ok. Update in progress.

randaz81, 2017-12-14:
I updated configuration files by adding missing parameters.

I was not able to test them on a real robot, so please test them carefully, keeping the red button pressed during the first launch and checking with the yarpmotorgui if the encoder values of the large joints are ok (it may require a couple of minutes for the timeouts to expire and complete the calibration). 

AxisName parameter in file icub_face.xml shoud be modified by you, replacing joint0,joint1 etc with the following strings: jaw, eyelids, lip-right, lip-left, lip-up, lip-bottom. The order in the file should be the same one displayed by yarpmotorgui (since I don't have a talking head robot here, I'm not sure about the order of the joints). This parameter, however, does not have any effect on the control.

An error message about motorPwmLimit is normal.

If yarprobotinterface does not startup because some other parameter is still missing, please tell me.

EliseiFrederic, 2017-12-15:
Thanks @barbalberto and @randaz81 for the update !
I'm ok with the robot-config pull + make install.
Not sure about the make install related to canbusmotioncontrol...

That's probably why we still can't start yarprobotinterface, though the previous errors
with axisname all disappeared.

It seems to complain about cfw2can : 
<...>
||| finding paths [plugins]
||| finding paths [plugins]
Cfw2Can::open() returning error -2
yarpdev: ***ERROR*** driver <cfw2can> was found but could not open
[ERROR]SharedCanBus: could not instantiate can device
yarpdev: ***ERROR*** driver <sharedcan> was found but could not open
[ERROR]Could not instantiate CAN device
[ERROR]Unable to unitialize CAN driver 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
[WARNING]Cannot open device left_arm_mc 
[WARNING]Cannot open device left_arm_mc 

Full file is attached : [log_test_yarprobotinterface_20171215.txt](https://github.com/robotology/icub-support/files/1562571/log_test_yarprobotinterface_20171215.txt)

Just in case it helps : 
$ yarp plugin --verbose cfw2can
Yes, this is a YARP plugin
Trying to find library '/usr/local/src/robot/yarp/build-pc104/lib/yarp/cfw2can.so' containing function 'cfw2can' -- fail
error while opening /usr/local/src/robot/yarp/build-pc104/lib/yarp/cfw2can.so:
  /usr/local/src/robot/yarp/build-pc104/lib/yarp/cfw2can.so: cannot open shared object file: No such file or directory
Trying to find library '/usr/local/src/robot/icub-main/build-pc104/lib/iCub/cfw2can.so' containing function 'cfw2can' -- found
  * library:        /usr/local/src/robot/icub-main/build-pc104/lib/iCub/cfw2can.so
  * system version: 5
  * class name:     yarp::dev::Cfw2Can
  * base class:     yarp::dev::DeviceDriver

I confirm I still have to include 
#include "/lib/modules/4.9.0-3-rt-amd64/iCubDrivers/cfw002/LinuxDriver/API/libc
instead of #include "libcfw002.h"
in src/libraries/icubmod/cfw2Can/Cfw2Can.{cpp,h} to be able to compile cfw2can

mbrunettini, 2017-12-15:
I'm trying to fix the issue working on a new icub-live release.
I'll ket you know soon

mbrunettini, 2017-12-18:
A [new release](http://www.icub.org/download/pc104-images/icub-live_7.9-17.12.18-rt-amd64.zip) of iCub LIVE is now available. We are still testing it, but it fixes the cfw002 module loading and cfw API library linking.

EliseiFrederic, 2017-12-20:
Indeed, the new version fixes the link/execution problem with the cfw2 problem. Thanks Matteo !

I now have the expected messages about our outofdate firmware : 
<...>
||| finding paths [plugins]
||| finding paths [plugins]
[INFO]created device <cfw2can>. See C++ class yarp::dev::Cfw2Can for documentation.
[WARNING]SharedCanBus [1] using default thread period = 10ms
[INFO]created device <sharedcan>. See C++ class yarp::dev::CanBusAccessPoint for documentation.
printing destinations and inverted map
0 1
2 2
4 3
6 4
[WARNING]Set baud rate not allowed from CanBusAccessPoint implementation
[INFO]using ICUB_CANMASKS_STRICT_FILTER option
[DEBUG]class 0 set
[DEBUG]class 1 set
[DEBUG]class 2 set
[DEBUG]class 3 set
[DEBUG]Can read/write buffers created, buffer size: 500
[INFO]CanBusResources::initialized correctly
Registering new thread 0 out of 500
[INFO]
[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.6
[INFO]Firmware report:
[INFO]sharedcan [1] joint: 0 can_address:  1 board type: 4 (BLL) version: 2.50 build: 83 CAN_protocol:1.3
[INFO]sharedcan [1] joint: 1 can_address:  1 board type: 4 (BLL) version: 2.50 build: 83 CAN_protocol:1.3
[INFO]sharedcan [1] joint: 2 can_address:  2 board type: 4 (BLL) version: 2.57 build: 83 CAN_protocol:1.3
[INFO]sharedcan [1] joint: 3 can_address:  2 board type: 4 (BLL) version: 2.57 build: 83 CAN_protocol:1.3
[INFO]sharedcan [1] joint: 4 can_address:  3 board type: 3 (4DC) version: 2.19 build: 83 CAN_protocol:1.3
[INFO]sharedcan [1] joint: 5 can_address:  3 board type: 3 (4DC) version: 2.19 build: 83 CAN_protocol:1.3
[INFO]sharedcan [1] joint: 6 can_address:  4 board type: 3 (4DC) version: 2.19 build: 83 CAN_protocol:1.3
[INFO]sharedcan [1] joint: 7 can_address:  4 board type: 3 (4DC) version: 2.19 build: 83 CAN_protocol:1.3
[INFO]**********************************
[INFO]
[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]  It has been detected that your control boards are not using the same
[ERROR]  CAN protocol used by iCubinterface. iCubInterface cannot continue.
[ERROR]  Please update your system (iCubInterface and/or your control board firmware.
[ERROR]  For further information please visit: http://wiki.icub.org/wiki/Firmware
[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open

EliseiFrederic, 2017-12-20:
@barbalberto, I think I'll need help to update the firmware...

I tried the GUI way, and the automated way : 
* With FirmwareUpdate, I can discover, select the identical board, but then I have no idea wich files should be sent to what ?
* With the automatic script, we have errors ! Maybe the automatic script has not been update for Grenoble01 ?

icub@pc104:/usr/local/src/robot$ icub-firmware-build/updateRobot.sh

----------------------------------------------
Welcome to the automatic firmware updater tool
----------------------------------------------

no filename specified, using default configuration...
cannot open the firmware description file in default directory, searching in alternate path...

found firmware description file:
/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/firmwareUpdate.txt

Remember: Never update the firmware if iCubInterface/robotInterface is running!
If you are not sure about the current status of the robot, turn off the motor boards, turn them on again and wait at 5 seconds before proceeding.
Do you want to proceed with the firmware update? (Y/N)
Y
canLoader --canDeviceType cfw2can --canDeviceNum 0 --boardId 1 --firmware 2BLLDC.1.62.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 0 --boardId 2 --firmware 4DC.2.15.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 0 --boardId 7 --firmware 4DC.2.15.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 0 --boardId 9 --firmware 4DC.1.11.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 0 --boardId 5 --firmware 2BLL_IE.2.54.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 0 --boardId 6 --firmware 2BLL_IE.2.52.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 1 --boardId 1 --firmware 2BLL_IE.2.50.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 1 --boardId 2 --firmware 2BLL_IE.2.57.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 1 --boardId 3 --firmware 4DC.2.19.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 5 --boardId 5 --firmware 4DC.2.28.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 5 --boardId 7 --firmware 4DC.2.30.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
canLoader --canDeviceType cfw2can --canDeviceNum 2 --boardId 1 --firmware 2BLL_IE.2.50.out.S
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
[... 37 lines cut ...]
Initializing prompt version of canLoader...
[ERROR]invalid --firmware parameter 
[ERROR]file not found
cfw2can 0 1 2BLLDC.1.62.out.S --> Error: INVALID_PARAM_FILE
cfw2can 0 2 4DC.2.15.out.S --> Error: INVALID_PARAM_FILE
cfw2can 0 7 4DC.2.15.out.S --> Error: INVALID_PARAM_FILE
cfw2can 0 9 4DC.1.11.out.S --> Error: INVALID_PARAM_FILE
cfw2can 0 5 2BLL_IE.2.54.out.S --> Error: INVALID_PARAM_FILE
cfw2can 0 6 2BLL_IE.2.52.out.S --> Error: INVALID_PARAM_FILE
cfw2can 1 1 2BLL_IE.2.50.out.S --> Error: INVALID_PARAM_FILE
cfw2can 1 2 2BLL_IE.2.57.out.S --> Error: INVALID_PARAM_FILE
cfw2can 1 3 4DC.2.19.out.S --> Error: INVALID_PARAM_FILE
cfw2can 5 5 4DC.2.28.out.S --> Error: INVALID_PARAM_FILE
cfw2can 5 7 4DC.2.30.out.S --> Error: INVALID_PARAM_FILE
cfw2can 2 1 2BLL_IE.2.50.out.S --> Error: INVALID_PARAM_FILE
cfw2can 2 2 2BLL_IE.2.57.out.S --> Error: INVALID_PARAM_FILE
cfw2can 2 3 4DC.2.19.out.S --> Error: INVALID_PARAM_FILE
cfw2can 6 5 4DC.2.28.out.S --> Error: INVALID_PARAM_FILE
cfw2can 6 7 4DC.2.30.out.S --> Error: INVALID_PARAM_FILE
cfw2can 3 5 2BLL_IE.2.51.out.S --> Error: INVALID_PARAM_FILE
cfw2can 3 6 2BLL_IE.2.51.out.S --> Error: INVALID_PARAM_FILE
cfw2can 3 7 2BLL_IE.2.51.out.S --> Error: INVALID_PARAM_FILE
cfw2can 4 8 2BLL_IE.2.51.out.S --> Error: INVALID_PARAM_FILE
cfw2can 4 9 2BLL_IE.2.51.out.S --> Error: INVALID_PARAM_FILE
cfw2can 4 10 2BLL_IE.2.51.out.S --> Error: INVALID_PARAM_FILE

EliseiFrederic, 2017-12-21:
updateRobot.sh worked sucessfully, when launched directly from icub-firmware-build :-)

May I suggest to modify : 
> Launch the firmware update script (icub-firmware-build/updateRobot.sh):
from http://wiki.icub.org/wiki/Firmware#Standard_firmware_update_.28DSP.29
into something like
"Launch the firmware update script updateRobot.sh from the icub-firmware-build directory" ?

EliseiFrederic, 2017-12-21:
So, I tried to launch yarprobotinterface with red button pressed, to validate the robotconfig files that @randaz81 updated.

Everything seems fine up to : 
<...>
[INFO]Entering action level 5 of phase startup 
[ERROR]Target device TalkingHead_Calibrator (network = Calibrator ) does not exist. 
[ERROR]Cannot run attach action on device head_mc_wrapper 
[ERROR]Target device TalkingFace_Calibrator (network = Calibrator ) does not exist. 
[ERROR]Cannot run attach action on device face_mc_wrapper 

This might be the cause that later it starts an automatic shutdown : 
<...>
[INFO]icub/cartesianController/right_arm started successfully
[INFO]All actions for action level 15 of startup phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 15 of startup phase finished. 
[WARNING]There was some problem running actions for startup phase . Please check the log and your configuration 
[INFO]startup phase finished. 
[ERROR]Error in startup phase... see previous messages for more info 
[WARNING]Interrupt # 1 # received. 
[INFO]Interrupt received. Stopping all running threads. 
[INFO]interrupt1 phase starting... 
[INFO]Entering action level 1 of phase interrupt1 
<...>

Any idea of the cause for this problem ?
Any .ini file than I can easily modify to check for the other parts of the robot ?

If that helps, I see 3 files in /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/calibrators : 
-rw-r--r-- 1 icub icub 2713 déc.   7 19:05 face_calib.xml
-rw-r--r-- 1 icub icub 2709 déc.   7 19:05 head_calib.xml
-rw-r--r-- 1 icub icub 3510 déc.   7 19:05 head_face_calib.xml
but 
icub_all.xml   included by yarpmotorgui.ini
only references :
    
    <devices file="calibrators/head_calib.xml" />
    <devices file="calibrators/face_calib.xml" />

barbalberto, 2017-12-21:
Hi @EliseiFrederic, there is a mismatch in the wrapper config file.

in https://github.com/robotology/robots-configuration/blob/master/iCubGrenoble01/wrappers/motorControl/face_mc_wrapper.xml @line 19 the keywork "Calibrator"  must refer to the tag `name` of the device `parametricCalibrator`, which in the file https://github.com/robotology/robots-configuration/blob/master/iCubGrenoble01/calibrators/face_calib.xml at line 4 is is set to "face_calibrator"

So, to cut a long story shirt, in hte wrapper files, change the "Calibrators" param to `head_calibrator` and `head_calibrator` respectively. I think that should fix the issue.

EliseiFrederic, 2017-12-21:
Thanks for this solution !
yarprobotinterface now stays running 👍 

I can launch yarpmotorgui, with red button pressed, in the hope to validate than the measured/corrected joints are in the correct range within the new robotconfig xml files.
The joint values are updated (encoders). But neither the "position" sliders, nor their range, are displayed (as they used too with the old motorgui). 
It seems they don't even get loaded : 

$ yarpmotorgui 
||| configuring
||| default config file specified as yarpmotorgui.ini
||| checking [/localdata/exports/code/robots-configuration/build/yarpmotorgui.ini] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/yarpmotorgui.ini] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/yarpmotorgui.ini
QObject::connect: No such signal MainWindow::internalClose()
QObject::connect:  (sender name:   'MainWindow')
QObject::connect:  (receiver name: 'MainWindow')
[WARNING]robotDescriptionServer not found, robot parts will be set manually. 
[DEBUG]Appending /icub/head
[DEBUG]Appending /icub/face
[DEBUG]Appending /icub/torso
[DEBUG]Appending /icub/left_arm
[DEBUG]Appending /icub/right_arm
[DEBUG]Appending /icub/right_leg
[DEBUG]Appending /icub/left_leg
[DEBUG]Checking the existence of: /yarpmotorgui0//icub/face/rpc:o 
[DEBUG]ADDRESS is:  
||| finding paths [plugins]
||| checking [/localdata/exports/code/robots-configuration/build/plugins] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/plugins] (robot)
||| checking [/home/icub/.config/yarp/plugins] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/plugins] (YARP_DATA_HOME)
||| checking [/etc/yarp/plugins] (YARP_CONFIG_DIRS)
||| checking [/share/yarp/plugins] (YARP_DATA_DIRS)
||| checking [/share/iCub/plugins] (YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins
||| checking [/usr/local/src/robot/icub-tests/suits/plugins] (YARP_DATA_DIRS)
yarp: Port /yarpmotorgui0//icub/face/rpc:o active at tcp://192.168.70.52:10078/
yarp: Port /yarpmotorgui0//icub/face/command:o active at tcp://192.168.70.52:10079/
yarp: Port /yarpmotorgui0//icub/face/stateExt:i active at tcp://192.168.70.52:10080/
yarp: Sending output from /yarpmotorgui0//icub/face/rpc:o to /icub/face/rpc:i using tcp
yarp: Sending output from /yarpmotorgui0//icub/face/command:o to /icub/face/command:i using udp
yarp: Receiving input from /icub/face/stateExt:o to /yarpmotorgui0//icub/face/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[DEBUG]Setting a valid finder 
yarp: Port /yarpmotorgui/face/sequence:o active at tcp://192.168.70.52:10081/
[DEBUG]Initializing interfaces...
[DEBUG]Opening interfaces...
[INFO]face iencs->getEncoders() ok!
[ERROR]Error while getting current range, part  face  joint  0 
[ERROR]Error while getting current range, part  face  joint  1 
[ERROR]Error while getting current range, part  face  joint  2 
[ERROR]Error while getting current range, part  face  joint  3 
[ERROR]Error while getting current range, part  face  joint  4 
[ERROR]Error while getting current range, part  face  joint  5 
[DEBUG]Checking the existence of: /yarpmotorgui0//icub/head/rpc:o 
[DEBUG]ADDRESS is:  
yarp: Port /yarpmotorgui0//icub/head/rpc:o active at tcp://192.168.70.52:10082/
yarp: Port /yarpmotorgui0//icub/head/command:o active at tcp://192.168.70.52:10083/
yarp: Port /yarpmotorgui0//icub/head/stateExt:i active at tcp://192.168.70.52:10084/
yarp: Sending output from /yarpmotorgui0//icub/head/rpc:o to /icub/head/rpc:i using tcp
yarp: Sending output from /yarpmotorgui0//icub/head/command:o to /icub/head/command:i using udp
yarp: Receiving input from /icub/head/stateExt:o to /yarpmotorgui0//icub/head/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[DEBUG]Setting a valid finder 
yarp: Port /yarpmotorgui/head/sequence:o active at tcp://192.168.70.52:10085/
[DEBUG]Initializing interfaces...
[DEBUG]Opening interfaces...
[INFO]head iencs->getEncoders() ok!
[ERROR]Error while getting current range, part  head  joint  0 
[... 119 lines cut ...]
[INFO]torso iencs->getEncoders() ok!
[ERROR]Error while getting current range, part  torso  joint  0 
[ERROR]Error while getting current range, part  torso  joint  1 
[ERROR]Error while getting current range, part  torso  joint  2 
yarp: Removing output from /yarpmotorgui0//icub/face/rpc:o to /icub/face/rpc:i
yarp: Removing output from /yarpmotorgui0//icub/face/command:o to /icub/face/command:i
yarp: Removing input from /icub/face/stateExt:o to /yarpmotorgui0//icub/face/stateExt:i
yarp: Removing output from /yarpmotorgui0//icub/left_arm/rpc:o to /icub/left_arm/rpc:i
yarp: Removing output from /yarpmotorgui0//icub/left_arm/command:o to /icub/left_arm/command:i
yarp: Removing input from /icub/left_arm/stateExt:o to /yarpmotorgui0//icub/left_arm/stateExt:i
yarp: Removing output from /yarpmotorgui0//icub/right_arm/rpc:o to /icub/right_arm/rpc:i
yarp: Removing output from /yarpmotorgui0//icub/right_arm/command:o to /icub/right_arm/command:i
yarp: Removing input from /icub/right_arm/stateExt:o to /yarpmotorgui0//icub/right_arm/stateExt:i
yarp: Removing output from /yarpmotorgui0//icub/torso/rpc:o to /icub/torso/rpc:i
[... cut ...]
yarp: Removing output from /yarpmotorgui0//icub/left_leg/rpc:o to /icub/left_leg/rpc:i
yarp: Removing output from /yarpmotorgui0//icub/left_leg/command:o to /icub/left_leg/command:i
yarp: Removing input from /icub/left_leg/stateExt:o to /yarpmotorgui0//icub/left_leg/stateExt:i
yarp: Removing output from /yarpmotorgui0//icub/head/rpc:o to /icub/head/rpc:i
yarp: Removing output from /yarpmotorgui0//icub/head/command:o to /icub/head/command:i
yarp: Removing input from /icub/head/stateExt:o to /yarpmotorgui0//icub/head/stateExt:i

EliseiFrederic, 2018-01-04:
Here is a snapshot of what we get...
How can we verify that the motor limits are ok with the new files ?
[image]

julijenv, 2018-01-05:
Hi @EliseiFrederic ,
if the head is positionned as if i would be already calibrated, the first 3 joints should be 0 or close (as the value that you can read are  coming from absolute encoder - so the value should reflect the position.
Could you give us a picture of the head position at starting? (how is the head in the rest position on which you put or left it).
from the value I would guess the head is bent in frontward position - chin close to the chest- Am i right?
for the other high value for joint 4 ( 5th joint of the head - eye_verison) you will need to launch the calibration for real in order to understand if everthing is alright becasue obth eye need to go to theextreme limit to determine where the zero is. (you may need later to adjust at yoour will by changing delta in the config files to be more precise).
let us knkw

EliseiFrederic, 2018-01-05:
You're probably perfectly right about the head orientation (it is left opened during the USB key upgrade process), but my question was more generic than just for the head as I was trying to follow advices from @randaz81 about the red button depressed for our first motor-try after the file updates : 
https://github.com/robotology/icub-support/issues/583#issuecomment-351783122

Should we *only validate that the 0 positions* for torso, shoulders etc are ok, *and not the min/max ranges* ? I remember that during the previous big upgrade of the robot with the older yarpmotorgui/file format I was asked to move the joints manually to check the ranges for the first run with the red button pressed.
I admit I'm relunctant to run without the red button, as the calibration movements could go too far without the right limits, couldn't they ?

Maybe it's a new behaviour of the release that *the limits are not read* if the red button is pressed ?
None of the encoders (torso, arms, legs...) could get their ranges, as posted in
https://github.com/robotology/icub-support/issues/583#issuecomment-353393990

[INFO]torso iencs->getEncoders() ok!
[ERROR]Error while getting current range, part torso joint 0
[ERROR]Error while getting current range, part torso joint 1
[ERROR]Error while getting current range, part torso joint 2

I understand that the calibration sort of alters the range and translate some hardware-read values of the joints into the software-displayed/used range. Maybe the newer/recent yarp/icub split does not mix these anymore and that's the reason why I get these errors and don't have them displayed, but I would prefer to have an explicit confirmation before taking the risk to run the robot without the red button ;-)

Thanks in advance !

julijenv, 2018-01-09:
Hi @EliseiFrederic ,
Marco @randaz81 , just want you to make sure that the values are close to zero when the robot is in a standing straight position (torso, leg and head straight like if the calibration phase has been done) - visual check before laucnhing fr real the robot. For the arms try to bring them up tp the final position of the calibration and check if the value are close to the usual one [ -30 30 0 45], check if everything looks alright - close the yarprobotinterface. at that point no need to fear anything just start the robot in the normal way ( NO RED BUTTON PRESSED) and have an eye during the first time the robot calibrates as  actuated and check if it goes to the position it should go.
If everything went smooth just control the movement of every single joint range and verify that it is good - but I would guess as those files mx min value are egual to all robots it wont be a problem anymore. 
Let us know

EliseiFrederic, 2018-01-16:
Hi, 
Torso, shoulders, face etc looked moving ok at the beginning of the startup procedure.
But the calibration fails and yarprobotinterface stops.
I think we have a trouble with the values/calibration of the wrist (staying outward, not moving).

Here is the log : 

icub@pc104:~$ yarprobotinterface 
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./icub_all.xml
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./icub_all.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./cartesian/left_arm_cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./cartesian/right_arm_cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/motorControl/left_arm_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/motorControl/right_arm_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/motorControl/left_leg_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/motorControl/right_leg_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/motorControl/head_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/motorControl/face_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/motorControl/torso_mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/motorControl/icub_left_arm.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/motorControl/icub_left_hand.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/motorControl/icub_right_arm.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/motorControl/icub_right_hand.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/motorControl/icub_left_leg.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/motorControl/icub_right_leg.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/motorControl/icub_head.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/motorControl/icub_face.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/motorControl/icub_torso.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/VFT/left_arm_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/VFT/left_leg_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/VFT/right_arm_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/VFT/right_leg_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./wrappers/VFT/torso_VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/VFT/left_arm_virtual_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/VFT/left_leg_virtual_strain.xml 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/./hardware/VFT/right_arm_virtual_strain.xml 
[... cut ...]
[DEBUG]sharedcan [0] msg from board 6: CMODE SET:1 
[DEBUG]sharedcan [3] msg from board 5: CMODE SET:1 
[DEBUG]sharedcan [3] msg from board 5: PWM ENA:1 
[DEBUG]sharedcan [3] msg from board 6: PWMEN CH0 
[DEBUG]sharedcan [3] msg from board 6: PWM ENA:1 
[DEBUG]sharedcan [3] msg from board 6: CMODE SET:1 
[DEBUG]sharedcan [3] msg from board 6: PWMEN CH1 
[DEBUG]sharedcan [3] msg from board 6: PWMEN A:0 
[DEBUG]sharedcan [3] msg from board 6: CMODE SET:1 
[DEBUG]sharedcan [3] msg from board 7: PWMEN CH0 
Segmentation fault

EliseiFrederic, 2018-01-16:
Thanks for your help, iCubGrenoble01 now compiles and run !

We still have **red errors** in yarprobotinterface and yarpmotorgui though : 

* in **yarprobotinterface** : 
[ERROR]TalkingFace_Calibrator :Timeout while calibrating  2 
[ERROR]TalkingFace_Calibrator : set 1 : Calibration went wrong! Disabling axes and keeping safe pid limit

* in **yarpmotorgui** for every of the joints groups (face, head, arm, legs and torso) : 
[DEBUG]Initializing interfaces...
[DEBUG]Opening interfaces...
[INFO]torso iencs->getEncoders() ok!
[ERROR]Error while getting current range, part  torso  joint  0 
[ERROR]Error while getting current range, part  torso  joint  1 
[ERROR]Error while getting current range, part  torso  joint  2 

though the sliders intervals in the GUI are displayed now and seem correct...

EliseiFrederic, 2018-01-31:
It takes more than 11 seconds to start yarpmotorgui, because of all the 
[ERROR]Error while getting current range, part  <XXX>

The pending messages in yarprobotinterface are : 
[ERROR]getCurrentRangeRaw not yet implemented for CanBusMotionControl

Anything we can do about this ?

barbalberto, 2018-02-01:
Some error messages about something missing is perfectly normal. The motorGui, at startup, asks a lot of stuff to the robot, some of them are implemented, some other aren't, depending on the robot. That one in particular is asking the maximum allowed current the motor can handle. That's a side information the gui can safely ignore, it is just a piece of information to be displayed.

The error about something missing piece of information does not have any impact on the start-up timing.

EliseiFrederic, 2018-02-01:
Thanks for your answer.
So 11 seconds to start yarpmotorgui is a normal timing ?
If yes, I think we can close this issue !

barbalberto, 2018-02-01:
Up to 5 or 6 second is normal, 11 is quite high, but it may be related to network issues or CPU workload. 
I simply meant it is not related to the `getSomeThing` errors. Just wait for the robotInterface startup to be completed before starting the gui. A part from that, I can't think of anything else to speed up the gui.

EliseiFrederic, 2018-02-05:
Maybe I was not clear : 
- yarprobotinterface has finished initialization. With no red error.
- everytime I start yarpmotorgui : yarpmotorgui needs 11 seconds, prints the red error messages on its own, and get yarprobotinterface to print his own message
- leaving yarpmotorgui and restarting it gives no improvement. starting a second one while the first is running does the same too.

traversaro, 2019-10-03:
> Thanks for your answer. So 11 seconds to start yarpmotorgui is a normal timing ? If yes, I think we can close this issue !

Hi @EliseiFrederic . I do not know if you are still able to reproduce this issue, but indeed has @barbalberto mentions that could be related to the computer on which yarpmotorgui is launched. However, I strongly suggest to open a new issue if that is still a problem, thanks! As the main problem reported (compilation failure) has been solved, I think we can close this issue.

closed 2019-10-03 by traversaro