robotology/mesh-support #585: The IMU measurements are constant for 100 milliseconds

traversaro, 2017-12-13:
### Description of the failure
The output of the IMU, that is supposed to be refreshed every 100 Hz, is sometimes constant for more then 100 milliseconds.

### Detailed conditions and logs of the failure
This is an example log of a dump obtained with `yarp read ... envelope /icub/inertial`:
~~~
10218 1513177141.53712 0.456266268049857 -41.0097163905489 16.4803043997419 7.62 -0.15 6.45 3.875 -0.375 5.5 -67.375 29.25 -133.375
10219 1513177141.54725 0.35954154038283 -41.0199868494101 16.9166935858489 6.72 -0.1 7.15 -1.4375 6.3125 -3.625 -66.25 29.0 -134.75
10220 1513177141.55739 0.35954154038283 -41.0199868494101 16.9166935858489 6.72 -0.1 7.15 -1.4375 6.3125 -3.625 -66.25 29.0 -134.75
10221 1513177141.5675 0.35954154038283 -41.0199868494101 16.9166935858489 6.72 -0.1 7.15 -1.4375 6.3125 -3.625 -66.25 29.0 -134.75
10222 1513177141.57761 0.35954154038283 -41.0199868494101 16.9166935858489 6.72 -0.1 7.15 -1.4375 6.3125 -3.625 -66.25 29.0 -134.75
10223 1513177141.5877 0.35954154038283 -41.0199868494101 16.9166935858489 6.72 -0.1 7.15 -1.4375 6.3125 -3.625 -66.25 29.0 -134.75
10224 1513177141.59781 0.35954154038283 -41.0199868494101 16.9166935858489 6.72 -0.1 7.15 -1.4375 6.3125 -3.625 -66.25 29.0 -134.75
10225 1513177141.60792 0.491810619395339 -40.6394381376426 16.6297370298622 5.35 0.13 7.93 -1.125 -0.25 -3.875 -65.875 29.6875 -135.0625
10226 1513177141.61801 0.491810619395339 -40.6394381376426 16.6297370298622 5.35 0.13 7.93 -1.125 -0.25 -3.875 -65.875 29.6875 -135.0625
10227 1513177141.62813 0.491810619395339 -40.6394381376426 16.6297370298622 5.35 0.13 7.93 -1.125 -0.25 -3.875 -65.875 29.6875 -135.0625
10228 1513177141.63824 0.491810619395339 -40.6394381376426 16.6297370298622 5.35 0.13 7.93 -1.125 -0.25 -3.875 -65.875 29.6875 -135.0625
10229 1513177141.64832 0.491810619395339 -40.6394381376426 16.6297370298622 5.35 0.13 7.93 -1.125 -0.25 -3.875 -65.875 29.6875 -135.0625
10230 1513177141.65839 0.491810619395339 -40.6394381376426 16.6297370298622 5.35 0.13 7.93 -1.125 -0.25 -3.875 -65.875 29.6875 -135.0625
10231 1513177141.66851 0.563867410644579 -40.6788685379094 16.3781928165615 6.42 0.12 7.55 2.6875 -1.9375 3.0 -65.1875 29.375 -134.6875
10232 1513177141.6786 0.563867410644579 -40.6788685379094 16.3781928165615 6.42 0.12 7.55 2.6875 -1.9375 3.0 -65.1875 29.375 -134.6875
10233 1513177141.68873 0.563867410644579 -40.6788685379094 16.3781928165615 6.42 0.12 7.55 2.6875 -1.9375 3.0 -65.1875 29.375 -134.6875
10234 1513177141.69886 0.563867410644579 -40.6788685379094 16.3781928165615 6.42 0.12 7.55 2.6875 -1.9375 3.0 -65.1875 29.375 -134.6875
10235 1513177141.70897 0.563867410644579 -40.6788685379094 16.3781928165615 6.42 0.12 7.55 2.6875 -1.9375 3.0 -65.1875 29.375 -134.6875
10236 1513177141.71908 0.563867410644579 -40.6788685379094 16.3781928165615 6.42 0.12 7.55 2.6875 -1.9375 3.0 -65.1875 29.375 -134.6875
10237 1513177141.72919 0.563867410644579 -40.6788685379094 16.3781928165615 6.42 0.12 7.55 2.6875 -1.9375 3.0 -65.1875 29.375 -134.6875
10238 1513177141.7393 0.513345099492128 -40.8426700472113 16.5936310013597 6.75 0.13 7.0 -0.875 -1.875 3.4375 -66.25 29.375 -134.75
10239 1513177141.74941 0.513345099492128 -40.8426700472113 16.5936310013597 6.75 0.13 7.0 -0.875 -1.875 3.4375 -66.25 29.375 -134.75
10240 1513177141.75953 0.513345099492128 -40.8426700472113 16.5936310013597 6.75 0.13 7.0 -0.875 -1.875 3.4375 -66.25 29.375 -134.75
10241 1513177141.76964 0.513345099492128 -40.8426700472113 16.5936310013597 6.75 0.13 7.0 -0.875 -1.875 3.4375 -66.25 29.375 -134.75
10242 1513177141.77978 0.513345099492128 -40.8426700472113 16.5936310013597 6.75 0.13 7.0 -0.875 -1.875 3.4375 -66.25 29.375 -134.75
10243 1513177141.78989 0.513345099492128 -40.8426700472113 16.5936310013597 6.75 0.13 7.0 -0.875 -1.875 3.4375 -66.25 29.375 -134.75
10244 1513177141.80002 0.274584020967103 -40.9424791461792 16.8312994966859 6.93 -0.73 6.97 0.9375 0.4375 -1.6875 -67.0 29.375 -134.75
10245 1513177141.81013 0.274584020967103 -40.9424791461792 16.8312994966859 6.93 -0.73 6.97 0.9375 0.4375 -1.6875 -67.0 29.375 -134.75
10246 1513177141.82021 0.274584020967103 -40.9424791461792 16.8312994966859 6.93 -0.73 6.97 0.9375 0.4375 -1.6875 -67.0 29.375 -134.75
10247 1513177141.8303 0.274584020967103 -40.9424791461792 16.8312994966859 6.93 -0.73 6.97 0.9375 0.4375 -1.6875 -67.0 29.375 -134.75
10248 1513177141.84041 0.274584020967103 -40.9424791461792 16.8312994966859 6.93 -0.73 6.97 0.9375 0.4375 -1.6875 -67.0 29.375 -134.75
10249 1513177141.85054 0.274584020967103 -40.9424791461792 16.8312994966859 6.93 -0.73 6.97 0.9375 0.4375 -1.6875 -67.0 29.375 -134.75
10250 1513177141.86066 0.497881837648964 -40.8212640612245 16.6437908490916 6.43 0.44 7.48 -0.125 2.5625 -2.625 -66.5625 29.0625 -135.6875
10251 1513177141.87075 0.497881837648964 -40.8212640612245 16.6437908490916 6.43 0.44 7.48 -0.125 2.5625 -2.625 -66.5625 29.0625 -135.6875
10252 1513177141.88087 0.497881837648964 -40.8212640612245 16.6437908490916 6.43 0.44 7.48 -0.125 2.5625 -2.625 -66.5625 29.0625 -135.6875
10253 1513177141.89093 0.497881837648964 -40.8212640612245 16.6437908490916 6.43 0.44 7.48 -0.125 2.5625 -2.625 -66.5625 29.0625 -135.6875
10254 1513177141.90105 0.497881837648964 -40.8212640612245 16.6437908490916 6.43 0.44 7.48 -0.125 2.5625 -2.625 -66.5625 29.0625 -135.6875
10255 1513177141.91116 0.497881837648964 -40.8212640612245 16.6437908490916 6.43 0.44 7.48 -0.125 2.5625 -2.625 -66.5625 29.0625 -135.6875
10256 1513177141.92129 0.497881837648964 -40.8212640612245 16.6437908490916 6.43 0.44 7.48 -0.125 2.5625 -2.625 -66.5625 29.0625 -135.6875
10257 1513177141.93141 0.584080244344624 -40.6730863314583 16.4688400460562 5.87 0.03 7.66 -1.875 -1.1875 0.1875 -66.5625 29.0 -134.6875
10258 1513177141.94154 0.584080244344624 -40.6730863314583 16.4688400460562 5.87 0.03 7.66 -1.875 -1.1875 0.1875 -66.5625 29.0 -134.6875
10259 1513177141.95159 0.584080244344624 -40.6730863314583 16.4688400460562 5.87 0.03 7.66 -1.875 -1.1875 0.1875 -66.5625 29.0 -134.6875
10260 1513177141.96164 0.584080244344624 -40.6730863314583 16.4688400460562 5.87 0.03 7.66 -1.875 -1.1875 0.1875 -66.5625 29.0 -134.6875
10261 1513177141.9667 0.584080244344624 -40.6730863314583 16.4688400460562 5.87 0.03 7.66 -1.875 -1.1875 0.1875 -66.5625 29.0 -134.6875
10262 1513177141.97679 0.584080244344624 -40.6730863314583 16.4688400460562 5.87 0.03 7.66 -1.875 -1.1875 0.1875 -66.5625 29.0 -134.6875
10263 1513177141.9869 0.496306447434364 -40.7685932788441 16.5063153090885 6.84 0.23 6.89 0.9375 -4.5 5.3125 -67.0 28.6875 -135.25
10264 1513177141.99699 0.496306447434364 -40.7685932788441 16.5063153090885 6.84 0.23 6.89 0.9375 -4.5 5.3125 -67.0 28.6875 -135.25
10265 1513177142.0071 0.496306447434364 -40.7685932788441 16.5063153090885 6.84 0.23 6.89 0.9375 -4.5 5.3125 -67.0 28.6875 -135.25
10266 1513177142.01716 0.496306447434364 -40.7685932788441 16.5063153090885 6.84 0.23 6.89 0.9375 -4.5 5.3125 -67.0 28.6875 -135.25
10267 1513177142.02727 0.496306447434364 -40.7685932788441 16.5063153090885 6.84 0.23 6.89 0.9375 -4.5 5.3125 -67.0 28.6875 -135.25
10268 1513177142.03738 0.496306447434364 -40.7685932788441 16.5063153090885 6.84 0.23 6.89 0.9375 -4.5 5.3125 -67.0 28.6875 -135.25
10269 1513177142.0475 0.496306447434364 -40.7685932788441 16.5063153090885 6.84 0.23 6.89 0.9375 -4.5 5.3125 -67.0 28.6875 -135.25
10270 1513177142.05761 0.306712205130711 -41.0089944688084 16.8604293621609 7.16 -0.33 6.92 0.6875 3.75 -1.375 -65.5625 29.6875 -134.375
10271 1513177142.06772 0.306712205130711 -41.0089944688084 16.8604293621609 7.16 -0.33 6.92 0.6875 3.75 -1.375 -65.5625 29.6875 -134.375
10272 1513177142.07783 0.306712205130711 -41.0089944688084 16.8604293621609 7.16 -0.33 6.92 0.6875 3.75 -1.375 -65.5625 29.6875 -134.375
10273 1513177142.08795 0.306712205130711 -41.0089944688084 16.8604293621609 7.16 -0.33 6.92 0.6875 3.75 -1.375 -65.5625 29.6875 -134.375
10274 1513177142.09806 0.306712205130711 -41.0089944688084 16.8604293621609 7.16 -0.33 6.92 0.6875 3.75 -1.375 -65.5625 29.6875 -134.375
10275 1513177142.10817 0.306712205130711 -41.0089944688084 16.8604293621609 7.16 -0.33 6.92 0.6875 3.75 -1.375 -65.5625 29.6875 -134.375
10276 1513177142.11829 0.394003424288535 -40.7876671951455 16.7331545335933 6.04 0.09 7.6 -1.6875 0.6875 -3.9375 -67.375 29.0 -133.0
10277 1513177142.1284 0.394003424288535 -40.7876671951455 16.7331545335933 6.04 0.09 7.6 -1.6875 0.6875 -3.9375 -67.375 29.0 -133.0
[... cut ...]
10548 1513177144.83738 0.519597158055051 -40.7776148972086 16.4606997128147 6.01 0.08 7.71 -0.875 -1.0 -0.5 -66.1875 30.0625 -134.1875
10549 1513177144.8475 0.519597158055051 -40.7776148972086 16.4606997128147 6.01 0.08 7.71
~~~

The measurement sometime remains constant for more that 10 samples, i.e. more than 100 milliseconds. 
I think this is related to the following issues: 
* https://github.com/robotology/icub-support/issues/402#issuecomment-324013762
* https://github.com/robotology/icub-support/issues/449
* https://github.com/robotology/icub-support/issues/497

cc @barbalberto @pattacini

traversaro, 2017-12-13:
cc @DanielePucci

DanielePucci, 2017-12-13:
@barbalberto @randaz81

fiorisi, 2018-01-04:
@barbalberto @randaz81 did you have time to have a look at this problem? We are trying to [compare the measurements with the new IMU](https://github.com/robotology-playground/xsensmt-yarp-driver/issues/4), but the sampling of the head IMU is not sufficient for a fair comparison.

DanielePucci, 2018-01-09:
@barbalberto 
Unfortunately this issue is currently blocking our activities for comparing measurements coming from the IMUs in the head and in the waist. 

Did you have time to take a look at this problem? Or may you give us an estimate for when this issue will be addressed? 

Thanks

barbalberto, 2018-01-09:
From the data posted here, it seems that it is quite common to have at least 6 subsequent reading with the same values.

The driver shall print a small report every 30 seconds telling if any reading errors occourred, like
```
errors while reading acceleration:
errors while reading gyroscope   :
errors while reading magnetometer:
errors while reading angles      :
```
Can you check if any error is notified?

DanielePucci, 2018-02-07:
@barbalberto 
Any news on this? 

It would be nice if you could [double check](https://en.wikipedia.org/wiki/Red_team) the errors you mentioned just to rule out that we did not do some mistakes while interpreting data

CC
@randaz81

barbalberto, 2018-02-07:
Actually, I was waiting for input from your side, as I asked for info on last post and got no answer or comment or any kind of feedback yet.
If any difficulties is found to copy&paste that portion of the device output I can help, just tell me, please.
We can check together on Monday morning, is it ok for you guys?

P.S.
I don't get what you meant with that link about 'red team'

barbalberto, 2018-02-16:
I did some measurement on timing of the IMU. Apparently it tooks around 15 ms to read each piece of information (15 for acceler, 15 for gyro, 15 for magn ecc) and yes, sometime it gets stuck for about 70ms.

What I did is to collapse all the readings in a single bigger read for all the data (and some garbage in between). This reduces the mean time of reading from about 60ms to about 20ms. A snapshot hereafter:
```
READ TIME  0.0159256 
READ TIME  0.0159538 
READ TIME  0.0159197 
READ TIME  0.01595 
READ TIME  0.0159581 
READ TIME  0.015877 
READ TIME  0.0159752 
READ TIME  0.0159149 
READ TIME  0.0159509 
READ TIME  0.0639012    <---
READ TIME  0.0159423 
READ TIME  0.0159039 
READ TIME  0.0159528 
READ TIME  0.0159492 
READ TIME  0.0159507 
READ TIME  0.0159316 
READ TIME  0.0159652 
READ TIME  0.0159225 
READ TIME  0.0159261 
READ TIME  0.0159566 
READ TIME  0.0159216 
READ TIME  0.015969 
READ TIME  0.0159452 
READ TIME  0.0159411 
```
But yes, sometimes the serial line gets stuck for 70ms and that's it.  It is a single system call, and there is no lower level to act onto. That simply means the device is busy and somehow cannot and answer at the moment.

Anyway, the update rate at yarp level should look faster now. Let me know

pattacini, 2018-02-16:
As it is then, this IMU is simply useless 👎 
We need to react fast. Can we organize a meeting next week?

barbalberto, 2018-02-16:
is 16 ms not enough?

pattacini, 2018-02-16:
data freeze is catastrophic

barbalberto, 2018-03-09:
@maggia80 proposed to use a IMU connected via i2c instead of a tty serial.
When the board will be available we can test it and see if it does not freeze.

DanielePucci, 2018-04-17:
@barbalberto @maggia80 
Any news on the tests with the board?

CC
@randaz81 @lornat75

maggia80, 2018-04-18:
It's my fault. I didn't prepare the cable for testing the IMU with I2C interface. I will prepare it for tomorrow.

maggia80, 2018-04-19:
I have done the cable and I have done few modification on the board:

Added a BAV99 to reduce voltage to 3.6V and remove R113

[image]

pattacini, 2018-06-04:
Any update on this?

maggia80, 2018-06-05:
I have done another modification to the board adding a 0R on R107. It was necessary for setting the I2C address of the sensor.  I know that @Nicogene managed to read from the IMU and he was working on the driver. The test has been done on the Kontron COM-Express, that is the new one.

Nicogene, 2018-06-05:
The driver was almost ready ~1 month ago, unfortunately YARP 3 release took all my time, after the release I will back on it ! :muscle:

Nicogene, 2018-06-12:
Device ready to be shipped  robotlogy/yarp#1743

closed 2018-10-26 by maggia80

traversaro, 2019-01-16:
If there are some non-i2c IMUs around, or in general if we have the same problem with other [FTDI](https://www.ftdichip.com/)-based serial drivers, it would be interesting to understand if we can set the latency timer to zero, see https://www.ftdichip.com/Support/Knowledgebase/index.html?settingacustomdefaultlaten.htm and https://www.ftdichip.com/Support/Knowledgebase/index.html?an232beffectbuffsizeandlatency.htm . 

The functionality is present in the Linux kernel driver for FTDI devices (see https://github.com/torvalds/linux/blob/903b77c631673eeec9e9114e9524171cdf9a2646/drivers/usb/serial/ftdi_sio.c#L1681) but I do not know if it is hooked in someway. This Ubuntu question may be relevant: https://projectgus.com/2011/10/notes-on-ftdi-latency-with-arduino/ .

 Another possible alternative could be to connect to FTDI from the user-space (using for example https://www.intra2net.com/en/developer/libftdi/), bypassing the kernel serial driver.

traversaro, 2019-01-16:
Discussing with @maggia80 it seems that the FTDI delay it was not the only problem with the serial-based communication with the BNO055, but the information about the FTDI latency timer can still be useful in other projects.