robotology/mesh-support #588: Install FT sensors with STRAIN2

DanielePucci, 2017-12-22:
We would need to install the FT sensors SN 292, 293, and 294, which have STRAIN2, in the `left_leg`, `right_foot`, and `left_foot`. These sensors with new electronics are necessary to avoid saturations during walking. 

I have given these FT sensors to @julijenv, who will most likely mount the sensors on Wednesday 27th. 

These sensors are not calibrated, so torque control on `iCubGenova04` will not be functioning till next calibration.

@gabrielenava @S-Dafarra @traversaro @nunoguedelha @prashanthr05 @fjandrad

fjandrad, 2017-12-22:
@nunoguedelha this are going to be the new sensors I told you about so the procedure should be the one you were doing the previous days. With the IITfriendly variable to true and the identity as calibration matrix since they have not being calibrated before. You can even leave the calibFlag to true in this case for both device and params.m file

@DanielePucci  @traversaro For the left leg there should be no problems, specially since nuno and @prashanthr05 calibrate a sensor in that position recently the proccess should be quicker now. What I am afraid is if the excitation in the feet would be enough considering that in the pole the mass the percieve is really small compared to the expected range of the sensor (at least the 330N of the weight of iCub). I had discussed with @fiorisi the possibility of attaching some known mass in the feet for tackling  this issue, but as of now we do not have a way of rigidly attaching a mass to the feet. 

The other concern is that currently we are using the yoga in feed forward as part of the calibration proccess. During this week it was neccesary to have at least one sensor on the feet to provide the switching threshold for the yoga. I do not know if it is possible to do it basing the threshold on the legs instead of the feet otherwise another solution for the switching is required. What do you think @gabrielenava ? Yoga is prefered over an static position because it seems to excite sufficiently the torques for the sensors in the leg.

DanielePucci, 2017-12-22:
Thanks a lot @fjandrad for taking an eye open on us from far away 💯 

You are indeed right: the excitation may not be enough for calibrating the feet FT sensors. 

What we have thinking of is to take the calibration sets **during walking experiments run in position control**. This  avoids the need of having FT sensors as a whole, and it also helps in having calibration data with robot motions close to the walking patterns we will want to stabilise with torque control.

What do you think?

fjandrad, 2017-12-22:
@DanielePucci It might be worth considering. Code might need some tinkering to consider an experiment with changing support contacts as part of the calibration data. This might be specific to the feet though, my impression on the sensors in the leg with current procedure is that it seemed to be enough (not considering the tz), but we have not explored what are the expected or "good" values when doing walking.

@nunoguedelha  also for got to mention. Try to ask accame if he can provide you with the sensors already with the gains changed before putting them in the robot. Otherwise, procedure and suggestd values can be found here [changing gains](https://github.com/loc2/component_ft-sensors/issues/2#issuecomment-347926190). The gains in the right leg are 16 and 36. This was already close to saturation values when walking, so a gain lower than 16 might be worth trying (10 maybe). I asked Accame for the required messages to obtain this gain before leaving hopefully they would be ready by now.

DanielePucci, 2017-12-22:
Thanks @fjandrad 
How much do you think that it is gonna take to change the code for handling contact switching ?

P.S.
I think that @marcoaccame reads 😄

DanielePucci, 2017-12-22:
BTW, I think we can discuss this in another task specific issue in [this](https://github.com/loc2/component_ft-sensors) repo

julijenv, 2017-12-27:
Hi @DanielePucci , @fjandrad , @nunoguedelha ,
It has been changed the 3 FT sensor given to me this morning.
Let me know if everything is alrgiht and please close the issue

julijenv, 2017-12-28:
Hi @DanielePucci , @fjandrad , @nunoguedelha , @traversaro ,
Just as a reminder the STRAIN have been mounted as this scheme
Left hip : S/N 292
Left ankle: S/N 293
Right Hip : S/N 282
Right ankle: S/N 294

fiorisi, 2017-12-29:
The new sensors (`left_leg`, `right_foot` and `left_foot`) have been tested and seems to measure "something" (have not been calibrated yet). 

[image]

[image]

closed 2017-12-29 by fiorisi